marcone/teslausb #415: Music files deleted on first sync

oaquique, 2020-01-25:
I installed teslausb-2020012 yesterday and after finalizing the set up process, connected teslausb back to my computer, found the expected CAM and MUSIC disk drives and copied my 26GB (6000+ files) music collection. I also had a few extra files in the TeslaMusic folder of my NAS, which the teslausb syncs against.

After connecting the new configured teslausb to my Model 3, all 6000+ music files were deleted on first sync and only those in the TeslaMusic NAS share were left. The Pushover alert I got reads:

Copied 16 music file(s), deleted 6311, skipped 99 previously-copied files, encountered 0 copy errors and 0 delete errors.

This is a new behavior as with the Beta 1 of this release (the september one) I was able to add files through the TeslaMusic share, but it won't delete my collection.

Any ideas?

marcone, 2020-01-25:
That behavior is not new, and unrelated to the image you used (the image contains no teslausb functionality; all the teslausb functionality is in the scripts that are downloaded at install/setup time).
It has been deleting files that are not present on the share since November 4 2019, so presumably you hadn't updated since at least then.

closed 2020-01-25 by marcone

oaquique, 2020-01-25:
Thanks for the clarification. I have not updated since the latest beta release (September) therefore I consider it a new feature for the January release. 😊

BTW, awesome job! Love teslausb!

Cheers,
Oswaldo