marcone/teslausb #419: wait_for_archive_to_be_unreachable and /tmp/archive_is_unreachable inconsistancy

pyjamasam, 2020-01-30:
The function `wait_for_archive_to_be_unreachable` doesn't behave the same way as `wait_for_archive_to_be_reachable` with regards to the testing files `/tmp/archive_is_reachable` and `/tmp/archive_is_unreachable`

Since `wait_for_archive_to_be_reachable` loops for every waiting for the archive to show up it checks for the file `/tmp/archive_is_reachable` each time through the loop.  This allows easy testing of the archive process.

`wait_for_archive_to_be_unreachable` does similar looping as `wait_for_archive_to_be_reachable` but it also is using the `retry` function to only loop for a set number of times.  Due to this it only actually goes through the top level loop once and exits before the test for `/tmp/archive_is_unreachable` is done.  So things always "time out".

In the end we get the same result.  The code moves on and we go back to waiting for the archive to show up again.

I wouldn't consider this a major issue, just more a consistency thing.  `wait_for_archive_to_be_unreachable` just needs to be re-written to do the retrying its self so that it can check for the presence of `/tmp/archive_is_unreachable` each time through the loop.

If I get a chance I'll try to submit a pull request with the changes.

marcone, 2020-02-02:
> `wait_for_archive_to_be_unreachable` does similar looping as `wait_for_archive_to_be_reachable` but it also is using the `retry` function to only loop for a set number of times. Due to this it only [...]

While the two functions are implemented differently, I don't think what you said here is correct. There is no logical difference between `if ! retry archive_is_reachable` and `if ! archive_is_reachable`, except that if the archive is currently not reachable, the `retry` version will take a bit longer. And if the archive is currently not reachable, there's no need to test for `/tmp/archive_is_unreachable`, and it's OK to exit right away. If the archive *is* reachable, it will check for `/tmp/archive_is_unreachable` and emulate it being not reachable.
Am I overlooking something?

pyjamasam, 2020-02-03:
I think what I was trying to get at was that the `/tmp/archive_is_unreachable` test was never actually executed (if the archive wasn't actually reachable).

You're right, it all works in the end, it was more that there was a little bit of inconsistency with regards to how the functions were implemented.

I only really noticed this cause I was working on a new archiver (or rather a modification to the existing rsync archiver to use the rsync protocol vs SSH) and I had things half baked and not totally working.

It was that "take a bit longer" that made me notice it.  If `/tmp/archive_is_unreachable` the code shouldn't have to wait.

I just figured I'd point it out and see if it was something that needed fixing/refactoring.  But you're right, other than the delay things do work just as expected.

I am going to close this issue as I don't think anything actually needs to be done.

closed 2020-02-03 by pyjamasam