marcone/teslausb #420: [RFE] Support for ext4 for camera footage

fatfurrycat, 2020-02-01:
Just been reading up on the 'Roadie' unit for Tesla dashcam. 
It appears to be very similar to TeslaUSB, and I suspect it probably uses a Pi Zero W to run on. 

One interesting thing is that it uses ext4 for the camera footage, which seems like a sensible idea, as it's much more resilient than FAT32, especially when you suddenly pull power. 

I understand that this makes things more difficult if you want to plug in to a Windows PC and pull footage, but I suspect most of us are actually sending footage to a NAS automatically, so it won't matter as much. 

Any chance of having this as an option during install?

marcone, 2020-03-11:
Should also add support for exFAT, since that's apparently supported by the car now.

raushel, 2020-04-11:
Apparently, ext4 is no longer working, even though it's still mentioned in the documentation:

https://www.tryroadie.com/blog/teslas-in-car-dashcam-file-system-changes-and-roadie-oh-my

marcone, 2026-01-20:
I won't be implementing this myself, but will consider a PR if someone wants to contribute one.