marcone/teslausb #423: Updates and Clip File Locations

GoodyGizmos, 2020-02-02:
It appears my videos are no longer syncing to my compute. I got back on here to see if I should update something. I see that @marcone has been making updates routinely, but I don't see a "How to run updates" section in the readme. How do I update this on my RPi?

Also, when I'm remoted (SSH) into the RPi and I've already remounted as RW, what folder can I check to see if the video clips actually recorded?

marcone, 2020-02-02:
> It appears my videos are no longer syncing to my computer. I got back on here to see if I should update something.
> 
> I see that @marcone has been making updates routinely, but I don't see a "How to install updates" section in the readme. How is this done?

https://github.com/marcone/teslausb/wiki

> Also, when I'm remoted (SSH) into the RPi and I've already remounted as RW, what folder can I check to see if the video clips actually recorded?

Remounting r/w has nothing to do with recording clips. I suggest looking at the /mutable/archiveloop.log to see why it's syncing anything

closed 2020-02-02 by marcone

GoodyGizmos, 2020-02-03:
@marcone Thank you. I ran the update instructions and rebooted, then checked the log you suggested. I looks like something is going on with the date/time jumping around (1 Feb to 3 Nov then 2 Feb). Also, what do you make of this (Dirty bit is set. Fs was not properly unmounted and some data may be corrupt)?

Sat  1 Feb 22:27:47 PST 2020: Waiting for archive to be unreachable...
==============================================
Thu  3 Nov 10:17:08 PDT 2016: Starting archiveloop...
Thu  3 Nov 10:17:09 PDT 2016: Waiting for time to be set by ntpd...
Sun  2 Feb 09:51:09 PST 2020: Time still not set, attempting to force it
Sun  2 Feb 09:51:09 PST 2020: Failed to set time
Sun  2 Feb 09:51:09 PST 2020: Ensuring music backing file is mounted...
Sun  2 Feb 09:51:09 PST 2020: Mounting /mnt/music...
Sun  2 Feb 09:51:09 PST 2020: Mounted /mnt/music.
Sun  2 Feb 09:51:09 PST 2020: Ensured music drive is mounted.
Sun  2 Feb 09:51:09 PST 2020: Running fsck on /mnt/music...
Sun  2 Feb 09:51:09 PST 2020: | fsck from util-linux 2.29.2
Sun  2 Feb 09:51:15 PST 2020: | fsck.fat 4.1 (2017-01-24)
Sun  2 Feb 09:51:15 PST 2020: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Sun  2 Feb 09:51:15 PST 2020: |  Automatically removing dirty bit.
Sun  2 Feb 09:51:15 PST 2020: | Performing changes.
Sun  2 Feb 09:51:15 PST 2020: | /dev/loop0: 4588 files, 1199326/1512855 clusters
Sun  2 Feb 09:51:15 PST 2020: Finished fsck on /mnt/music.
Sun  2 Feb 09:51:15 PST 2020: Unmounting /mnt/music...
Sun  2 Feb 09:51:15 PST 2020: Unmounted /mnt/music.
Sun  2 Feb 09:51:15 PST 2020: Archiving...
Sun  2 Feb 09:51:15 PST 2020: Disconnecting usb from host...
Sun  2 Feb 09:51:15 PST 2020: Disconnected usb from host.
Sun  2 Feb 09:51:15 PST 2020: Ensuring cam archive is mounted...
Sun  2 Feb 09:51:15 PST 2020: /mnt/archive is already mounted.
Sun  2 Feb 09:51:15 PST 2020: Ensured cam archive is mounted.
Sun  2 Feb 09:51:15 PST 2020: Ensuring music archive is mounted...
Sun  2 Feb 09:51:15 PST 2020: /mnt/musicarchive is already mounted.
Sun  2 Feb 09:51:15 PST 2020: Ensured music archive is mounted.
Sun  2 Feb 09:51:15 PST 2020: Starting recording archiving...
Sun  2 Feb 09:51:15 PST 2020: Sending IFTTT message for moved files.
Sun  2 Feb 09:51:16 PST 2020: Ensuring cam file is mounted...
Sun  2 Feb 09:51:16 PST 2020: Mounting /mnt/cam...
Sun  2 Feb 09:51:16 PST 2020: Mounted /mnt/cam.
Sun  2 Feb 09:51:16 PST 2020: Ensured cam file is mounted.
Sun  2 Feb 09:51:17 PST 2020: Running fsck on /mnt/cam...
Sun  2 Feb 09:51:17 PST 2020: | fsck from util-linux 2.29.2
Sun  2 Feb 09:51:24 PST 2020: | fsck.fat 4.1 (2017-01-24)
Sun  2 Feb 09:51:24 PST 2020: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Sun  2 Feb 09:51:24 PST 2020: |  Automatically removing dirty bit.
Sun  2 Feb 09:51:24 PST 2020: | Performing changes.
Sun  2 Feb 09:51:24 PST 2020: | /dev/loop0: 4433 files, 3209958/3406677 clusters

GoodyGizmos, 2020-02-09:
@marcone Any suggestions based on those logs?

marcone, 2020-02-09:
It doesn't look like you actually updated your Pi: that "Starting recording archiving..." message in the log was changed to include the number of files and folders it's going to archive in September 2019, so your setup is quite old.