marcone/teslausb #424: Samba on Android

hitnrun30, 2020-02-05:
Ok, I have my Pi zero setup and it is syncing with my server, but I am having the worst time getting to look at the files on the device if I connect to it through Samba.  First what app would you say is preferred.  Second how do I log in.  I changed both the pi and root user password but no matter what I do it wont let me log into the server.

marcone, 2020-02-06:
I haven't found an Android app to access SMB shares that I'm really happy with. They're all pretty slow it seems.
As for logging in, the default samba user/pass is pi/raspberry, same as for ssh login, unless you selected to enable guest mode, in which case no credentials should be needed (you generally would select "guest mode" in the client app, though I think you can also use guest/guest as the user/pass).
Changing the login password doesn't change the samba password. To change the samba password, use `smbpasswd`.

closed 2020-02-06 by marcone

hitnrun30, 2020-02-06:
wait is that the password in the one step conf.  I downloaded many Samba apps and put in pi/(password)  and even root/(password)  per your readme and set both and it will not let me log into the share.  I think it may just be easy to just set guest and then I will see the files.

marcone, 2020-02-06:
> wait is that the password in the one step conf.

No. There is no way to set the Pi's Samba password in the .conf file.