marcone/teslausb #427: Scheduled Reboot

ClanLee, 2020-02-10:
I've noticed that on my RP Zero, after a few days, the raspberry pi stops uploading videos.  I can log onto the pi using ssh and when I attempt to reboot it, it usually hangs.  The only fix is to unplug it from the car and plug it back in again.  Then the cycle continues.

I know we can use cron to schedule a reboot but I was wondering if it's possible for TeslaUSB to check to see if inactive, then perform a scheduled reboot at a specified time.

marcone, 2020-02-12:
If it stops working but you can still ssh in to it, it would be more useful to have a look around and see what the problem is, so that can be fixed.
If as you say it hangs when rebooting, wouldn't an automatic reboot hang too?

closed 2020-02-12 by marcone

ClanLee, 2020-02-13:
I've already replaced the SD card, the Raspberry Pi Zero and also the USB cable.  After 3, 4 or 5 days, I've noticed the pi stops uploading and when I attempt a software reboot, it hangs.  If I reboot via SSH within a day or two of the hardware reboot, it reboots without issues.

By rebooting on a nightly basis, I was hoping to prevent this from happening.