marcone/teslausb #433: Recovering cam files outside of the 10min save window

martycastilla, 2020-02-15:
I've looked on various sites for a solution, but I have been unsuccessful. So at the risk of being flamed for asking here,  here goes. 

My M3 was side swiped today and I forgot to hit the download icon until I was home. My Pi sync'd when I got home but the needed clips aren't there. Are there additional cam clips that are not transferred that I can access on the Pi. I read that with a USB drive you can access all the clips with a script. Is that possible here or am I hosed?

marcone, 2020-02-15:
Sorry to hear about your car. It may be possible to retrieve older clips.
How did you configure your Pi in terms of cam size, samba, snapshots, etc?

martycastilla, 2020-02-15:
Here is my .conf file that I used. is this what you need?

# Default variables for CIFS (Windows/Mac file sharing) setup
export ARCHIVE_SYSTEM=cifs
export archiveserver=10.0.0.3
export sharename=Tesla/TeslaCam
export shareuser=tesla
export sharepassword=model3
# export cifs_version="3.0"
export camsize=80%

# if you set camsize to less than 100%, but don't want music to use
# all the remaining space, you can specify how much space you want
# music to use by setting this variable
#export musicsize=10G

# If you want to automatically copy music from a CIFS share every time
# the Pi connects to wifi, set the following variable. The share is
# assumed to exist on the same server as the archive share. It can
# be the same share as the share used for backing up recordings, but
# the folder needs to be different.
export musicsharename=Tesla/TeslaMusic

# Wifi setup information. Note that Raspberry Pi Zero W only supports 2.4 GHz wifi
export SSID='WiFi'
export WIFIPASS='*********'

# If doing a headless (i.e. automatic) setup
export HEADLESS_SETUP=true

# uncomment to enable a samba server that exports the recordings
#export SAMBA_ENABLED=true
# uncomment to enable guest access to the samba server
#export SAMBA_GUEST=true

# uncomment to enable teslausb to act as a wifi access point with the given
# SSID, so you can access it while on the road.
export AP_SSID='TESLAUSB WIFI'
# Change this! The setup script will not accept the default 'password'
# Also note that the wifi password must be at least 8 characters.
export AP_PASS='**********'
# IP address is optional. The AP will give its clients IP addresses in the
# x.x.x.100-150 range so make sure you don't put the AP's IP address in
# that range.
export AP_IP='192.168.66.1'

# uncomment if you want to set the time zone to something other than the default 'BST'
# Can be an actual timezone, or "auto" to attempt automatic timezone detection
export timezone="America/Los_Angeles"

# By default there is a 20 second delay between connecting to wifi and
# starting the archiving of recorded clips. Uncomment this to change
# the duration of that delay
# export archivedelay=20

# Uncomment if you want to override the default hostname of "teslausb"
# export TESLAUSB_HOSTNAME=teslausb-Model3

# Uncomment if setting up Pushover push notifications
# export pushover_enabled=false
# export pushover_user_key=user_key
# export pushover_app_key=app_key

# Uncomment if setting up Gotify push notifications
# export gotify_enabled=false
# export gotify_domain=https://gotify.domain.com
# export gotify_app_token=put_your_token_here
# export gotify_priority=5

# Uncomment if setting up IFTTT push notifications
# export ifttt_enabled=false
# export ifttt_event_name=put_your_event_name_here
# export ifttt_key=put_your_key_here

# Uncomment if setting up AWS SNS notifications
# export sns_enabled=true
# export aws_region=us-east-1
# export aws_access_key_id=put_your_accesskeyid_here
# export aws_secret_key=put_your_secretkey_here
# export aws_sns_topic_arn=put_your_sns_topicarn_here

# TeslaUSB can optionally use the Tesla API to keep your car awake, so it can
# power the Pi long enough for the archiving process to complete. To enable
# that, please provide your Tesla account email and password below.
# TeslaUSB will only send your credentials to the Tesla API itself.
export tesla_email=*********
export tesla_password=********
# Please also provide your vehicle's VIN, so TeslaUSB can keep the correct
# vehicle awake.
export tesla_vin=5YJ3E1EBOJF*******

# Uncomment and change if you want setup scripts to be pulled
# from a different repo than github.com/marcone/teslausb
# export REPO=marcone

# Uncomment and change if you want a different branch than main-dev
# export BRANCH=main-dev

martycastilla, 2020-02-15:
In the backingfiles SD partition there is the snapshots folder with more snapshot folders inside each with and snap.bin, snap.bin.toc and an empty mnt folder.

marcone, 2020-02-15:
If you're lucky there's still a snapshot with "RecentClips" of the incident.
Look in /backingfiles/TeslaCam/

closed 2020-02-15 by marcone

martycastilla, 2020-02-15:
I copied them to another drive, but they won't play. These are straight MP4 files right?

marcone, 2020-02-15:
yes, normal mp4 files

martycastilla, 2020-02-15:
is backingfiles suppose to be an XFS partition?

martycastilla, 2020-02-15:
I figured out my problem. I was trying to pull the files directly from the SD via a card reader. I then saw the files are symlinks and realized the rpi has to running for the symlinks to work. I've ssh's into the rpi and started an scp to my NAS. The first file just completed and i can play it. I was so pissed off about what happen I obviously couldn't think clearly. Thank you for the jump start and this project.