marcone/teslausb #434: Unable to see videos when going through mobile app

kevinbright, 2020-02-16:
I have this apparently working, but I cannot see any video files while going through SMB via a mobile app on android.  I am not interested in CIFS backup aspects, so I just set that to 'none' and commented out the other lines.  I only want to be able to access the recordings via a mobile app connected to the TESLAUSB WIFI network.  As I said, everything seems to work fine.  When I mount it in the car, the car reads it as a driver and writes to it.  I can navigate via SMB on a file explorer app on my phone and get through the various folders.  It is creating folders correctly, but when I click on a folder, the folder shows as empty.  When I plug the Pi into a computer and wait a minute or two, it pops up as a drive and all of the folders are there with the videos correctly in each folder, so the videos are there, but I just cannot access them via a mobile app.   I've tried three mobile apps and all three can connect to the Pi and see the folders, but none can see what's in each folder.  What am I doing wrong?

marcone, 2020-02-16:
https://github.com/marcone/teslausb/issues/35

kevinbright, 2020-02-16:
I got it working.  I set the filesize to 32G rather than 64G (using a 128G sd card) and now it's working.

closed 2020-02-19 by marcone