marcone/teslausb #439: Re-running setup doesn't take AP changes from conf file

themonomers, 2020-03-03:
When I setup the access point and re-ran the setup using:
sudo -i
/root/bin/remountfs_rw
/root/bin/setup-teslausb selfupdate
/root/bin/setup-teslausb
reboot

I made a mistake on the AP_SSID name so I updated the config file and re-ran the commands above again.  It doesn't seem to change after that.  In fact, I even commented out the AP lines and re-ran it and it didn't change or deactivate like I thought it would. 

I was able to go into /etc/hostapd/hostapd.conf and change the SSID and reboot to fix it but I'd like to know if I did something wrong above - I thought editing the conf file and running setup-teslausb would reconfigure everything according to the updated conf file, no?

By the way, this is an awesome feature!

marcone, 2020-03-03:
That's expected. Some parts of setup are only done once.

closed 2020-03-03 by marcone

softcoder, 2020-03-03:
> When I setup the access point and re-ran the setup using:
> sudo -i
> /root/bin/remountfs_rw
> /root/bin/setup-teslausb selfupdate
> /root/bin/setup-teslausb
> reboot
>
> I made a mistake on the AP_SSID name so [...]

I found the exact same issue with the same fix

softcoder, 2020-03-03:
Can that be documented in the FAQ?