marcone/teslausb #440: location of USB files that are shared to the car

jorythompson, 2020-03-04:
I am trying to create a web interface to the files created by the car.  

I have searched through the code, but it is not clear to me where they are.  Specifically, I want to be able to browse these files while driving, I just cannot find the folder

thanks in advance

marcone, 2020-03-04:
There is no such folder, because obviously the car and the Pi cannot both be accessing the same file system at the same time.

closed 2020-03-04 by marcone

jorythompson, 2020-03-05:
I'm not really that familiar with the pi and how usb is implemented, but it seems that this project is able to allow memory to be accessed by the usb client as well as the pi itself: https://magpi.raspberrypi.org/articles/pi-zero-w-smart-usb-flash-drive

marcone, 2020-03-05:
That does more or less the same thing as teslausb, except they appear to be leaving the drive mounted to the Pi and the connected host at the same time, which will almost certainly lead to file system corruption, because neither computer is aware that another computer is also accessing the disk at the same time.

Can you explain a bit more about what you actually want to do? There may be other ways.

jorythompson, 2020-03-05:
I have some code I've written that provides a web sever that serves mp4
files so that you can view them in your browser. Currently I am pointing to
the back store folders, which are stale by design.
Once that is working with the correct folder, I'd like to be able to view
them on the car's screen by connecting to pi's network (like I currently do
with my phone and stale videos).

I would like to think that if you mount the USB as read-only from the pi,
the files will not be corrupted.

marcone, 2020-03-05:
> I have some code I've written that provides a web sever that serves mp4 files so that you can view them in your browser. Currently I am pointing to the back store folders, which are stale by design. 
Once that is working with the correct folder, I'd like to be able to view them on the car's screen by connecting to pi's network (like I currently do with my phone and stale videos).

The car will not connect to private/reserved network addresses, so this will not work without additional hacks to make the Pi appear to the car as if it is on the public internet.

> I would like to think that if you mount the USB as read-only from the pi, the files will not be corrupted.

They won't be corrupted, but the Pi will not be expecting the file system to change out from under it, so it might think things are corrupted when there's a mismatch between what it has cached and what is actually on the file system. That's why I added the snapshot mechanism.

jorythompson, 2020-03-05:
> I have some code I've written that provides a web sever that serves mp4
> files so that you can view them in your browser. Currently I am pointing to
> the back store folders, which are stale by design. [...]
I've read that, but how does it connect to home networks?

> I would like to think that if you mount the USB as read-only from the pi,
> the files will not be corrupted.
>
> They won't be corrupted, but the Pi will not be expecting the file system
> to change out [...]
Can we do both?

marcone, 2020-03-05:
> I've read that, but how does it connect to home networks?

It doesn't. It connects to your home wifi, but will not connect to any of the hosts on your network.

> Can we do both?

I don't know what you mean by that.

jorythompson, 2020-03-06:
Is it possible to open the USB drive as read-only, but when you get ready
to transfer the files, re-open it as RW (like you currently are doing)?

jorythompson, 2020-03-28:
Is it possible to open the USB drive as read-only, but when you get ready
to transfer the files, re-open it as RW (like you currently are doing)?
I'd really like to get this to work, if at possible.

marcone, 2020-03-28:
Why? What are you actually trying to achieve?

jorythompson, 2020-03-28:
I am trying to set up a web server that I can look at the videos while
still on the road.  I was recently in an accident and I had to wait to get
home to sift through the videos.  It was too cumbersome to do it with VCS
on my phone.

marcone, 2020-03-28:
Any reason you can't use snapshots, similar to what teslausb does for Samba access?
It's certainly possible to mount the drive image read-only, but to the Pi that would look like a filesystem that randomly has its data change out from under it, in other words it would look like a flaky/unreliable drive.

jorythompson, 2020-03-28:
perhaps that may work... How often are the snapshots updated?

marcone, 2020-03-28:
A new snapshot is taken every hour, but you can take one whenever you want. Samba takes one whenever someone connects, to provide an up to date view.

jorythompson, 2020-04-04:
Looks like this is going to be obe:
https://insideevs.com/news/407818/tesla-sentry-mode-infotainment-viewing/

Poncherelly, 2020-04-04:
This is good news. The teslausb continues to be useful however as it will copy the files to a share on your home server for posting later :)

Richard

Purchasing a TESLA?  Please use My Referral Code<https://ts.la/richard48613> for free supercharging.

Reply-to: reply@reply.github.com
Cc: subscribed@noreply.github.com

https://insideevs.com/news/407818/tesla-sentry-mode-infotainment-viewing/