marcone/teslausb #445: Newbie, please be kind :)  This is my log.  TeslaCam folder is on SD card but nothing records..

mhuk01, 2020-03-07:
pi@teslausb:~ $ /boot/teslausb-headless-setup.log
/boot/teslausb-headless-setup.log: line 1: Sat: command not found
/boot/teslausb-headless-setup.log: line 2: Sat: command not found
/boot/teslausb-headless-setup.log: line 3: Sat: command not found
/boot/teslausb-headless-setup.log: line 4: Sat: command not found
/boot/teslausb-headless-setup.log: line 5: Sat: command not found
/boot/teslausb-headless-setup.log: line 6: Sat: command not found
/boot/teslausb-headless-setup.log: line 7: Sat: command not found
/boot/teslausb-headless-setup.log: line 8: Sat: command not found
/boot/teslausb-headless-setup.log: line 9: Sat: command not found
/boot/teslausb-headless-setup.log: line 10: Sat: command not found
/boot/teslausb-headless-setup.log: line 11: Sat: command not found
/boot/teslausb-headless-setup.log: line 12: Sat: command not found
/boot/teslausb-headless-setup.log: line 13: Sat: command not found
/boot/teslausb-headless-setup.log: line 14: Sat: command not found
/boot/teslausb-headless-setup.log: line 15: Sat: command not found
/boot/teslausb-headless-setup.log: line 16: Sat: command not found
/boot/teslausb-headless-setup.log: line 17: Sat: command not found
/boot/teslausb-headless-setup.log: line 18: Fri: command not found
/boot/teslausb-headless-setup.log: line 19: Fri: command not found
/boot/teslausb-headless-setup.log: line 20: Fri: command not found
/boot/teslausb-headless-setup.log: line 21: Fri: command not found
/boot/teslausb-headless-setup.log: line 22: Fri: command not found
/boot/teslausb-headless-setup.log: line 23: Fri: command not found
/boot/teslausb-headless-setup.log: line 24: Fri: command not found
/boot/teslausb-headless-setup.log: line 25: Fri: command not found
/boot/teslausb-headless-setup.log: line 26: Fri: command not found
/boot/teslausb-headless-setup.log: line 27: Fri: command not found
/boot/teslausb-headless-setup.log: line 28: Fri: command not found
/boot/teslausb-headless-setup.log: line 29: Fri: command not found
/boot/teslausb-headless-setup.log: line 30: Fri: command not found
/boot/teslausb-headless-setup.log: line 31: Fri: command not found
/boot/teslausb-headless-setup.log: line 32: Fri: command not found
/boot/teslausb-headless-setup.log: line 33: Fri: command not found
/boot/teslausb-headless-setup.log: line 34: Fri: command not found
/boot/teslausb-headless-setup.log: line 35: Fri: command not found
/boot/teslausb-headless-setup.log: line 36: Fri: command not found
/boot/teslausb-headless-setup.log: line 37: Fri: command not found
/boot/teslausb-headless-setup.log: line 38: Fri: command not found
/boot/teslausb-headless-setup.log: line 39: Fri: command not found
/boot/teslausb-headless-setup.log: line 40: Fri: command not found
/boot/teslausb-headless-setup.log: line 41: Fri: command not found
/boot/teslausb-headless-setup.log: line 42: Fri: command not found
/boot/teslausb-headless-setup.log: line 43: Fri: command not found
/boot/teslausb-headless-setup.log: line 44: Fri: command not found
/boot/teslausb-headless-setup.log: line 45: Fri: command not found
/boot/teslausb-headless-setup.log: line 46: Fri: command not found
/boot/teslausb-headless-setup.log: line 47: Fri: command not found
/boot/teslausb-headless-setup.log: line 48: Fri: command not found
/boot/teslausb-headless-setup.log: line 49: Fri: command not found
/boot/teslausb-headless-setup.log: line 50: Fri: command not found
/boot/teslausb-headless-setup.log: line 51: Fri: command not found
/boot/teslausb-headless-setup.log: line 52: Fri: command not found
/boot/teslausb-headless-setup.log: line 53: syntax error near unexpected token `('
/boot/teslausb-headless-setup.log: line 53: `Fri  6 Mar 21:55:58 GMT 2020 : Modifying partition table for mutable (writable) partition for script usage...'

marcone, 2020-03-07:
No, that's not your log. That's what you get when try to run the log as if it were a script.
You want to run `cat /boot/teslausb-headless-setup.log`

closed 2020-03-07 by marcone

mhuk01, 2020-03-07:
pi@teslausb:~ $ cat /boot/teslausb-headless-setup.log
Sat 15 Jun 22:27:14 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sat 15 Jun 22:27:14 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Sat 15 Jun 22:27:15 BST 2019 : Rebooting...
Sat 15 Jun 22:27:43 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sat 15 Jun 22:27:43 BST 2019 : Grabbing main setup file.
Sat 15 Jun 22:27:45 BST 2019 : Starting setup.
Sat 15 Jun 22:27:45 BST 2019 : adding country code to wpa_supplicant.conf
Sat 15 Jun 22:27:45 BST 2019 : rebooting for changes to take effect
Sat 15 Jun 22:28:11 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sat 15 Jun 22:28:11 BST 2019 : Starting setup.
Sat 15 Jun 22:28:12 BST 2019 : reading config from /root/teslausb_setup_variables.conf
Sat 15 Jun 22:28:12 BST 2019 : curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev
Sat 15 Jun 22:28:12 BST 2019 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Sat 15 Jun 22:28:16 BST 2019 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Sat 15 Jun 22:28:19 BST 2019 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Sat 15 Jun 22:28:23 BST 2019 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Sat 15 Jun 22:28:26 BST 2019 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Fri  6 Mar 21:54:30 GMT 2020 : curl -o /tmp/setup-teslausb https://raw.githubusercontent.com/marcone/teslausb/f6e0bead51b652b2c1f2b95f6ab6b60653631afd/setup/pi/setup-teslausb
Fri  6 Mar 21:54:31 GMT 2020 : Downloaded /tmp/setup-teslausb ...
Fri  6 Mar 21:54:31 GMT 2020 : /root/bin/setup-teslausb is up to date
Fri  6 Mar 21:54:31 GMT 2020 : Updating package index files...
Fri  6 Mar 21:55:40 GMT 2020 : Verifying that the requested configuration is valid...
Fri  6 Mar 21:55:40 GMT 2020 : curl -o /tmp/verify-configuration.sh https://raw.githubusercontent.com/marcone/teslausb/f6e0bead51b652b2c1f2b95f6ab6b60653631afd/setup/pi/verify-configuration.sh
Fri  6 Mar 21:55:41 GMT 2020 : Downloaded /tmp/verify-configuration.sh ...
Fri  6 Mar 21:55:41 GMT 2020 : usb_drive is not set. SD card will be used.
Fri  6 Mar 21:55:41 GMT 2020 : Verifying that there is sufficient space available on the MicroSD card...
Fri  6 Mar 21:55:41 GMT 2020 : There is sufficient space available.
Fri  6 Mar 21:55:45 GMT 2020 : Downloading additional setup scripts.
Fri  6 Mar 21:55:45 GMT 2020 : curl -o /root/bin/remountfs_rw https://raw.githubusercontent.com/marcone/teslausb/f6e0bead51b652b2c1f2b95f6ab6b60653631afd/run/remountfs_rw
Fri  6 Mar 21:55:46 GMT 2020 : Downloaded /root/bin/remountfs_rw ...
Fri  6 Mar 21:55:46 GMT 2020 : curl -o /root/bin/make_snapshot.sh https://raw.githubusercontent.com/marcone/teslausb/f6e0bead51b652b2c1f2b95f6ab6b60653631afd/run/make_snapshot.sh
Fri  6 Mar 21:55:47 GMT 2020 : Downloaded /root/bin/make_snapshot.sh ...
Fri  6 Mar 21:55:47 GMT 2020 : curl -o /root/bin/mount_snapshot.sh https://raw.githubusercontent.com/marcone/teslausb/f6e0bead51b652b2c1f2b95f6ab6b60653631afd/run/mount_snapshot.sh
Fri  6 Mar 21:55:48 GMT 2020 : Downloaded /root/bin/mount_snapshot.sh ...
Fri  6 Mar 21:55:48 GMT 2020 : curl -o /root/bin/mount_image.sh https://raw.githubusercontent.com/marcone/teslausb/f6e0bead51b652b2c1f2b95f6ab6b60653631afd/run/mount_image.sh
Fri  6 Mar 21:55:49 GMT 2020 : Downloaded /root/bin/mount_image.sh ...
Fri  6 Mar 21:55:49 GMT 2020 : curl -o /root/bin/release_snapshot.sh https://raw.githubusercontent.com/marcone/teslausb/f6e0bead51b652b2c1f2b95f6ab6b60653631afd/run/release_snapshot.sh
Fri  6 Mar 21:55:49 GMT 2020 : Downloaded /root/bin/release_snapshot.sh ...
Fri  6 Mar 21:55:49 GMT 2020 : curl -o /root/bin/force_sync.sh https://raw.githubusercontent.com/marcone/teslausb/f6e0bead51b652b2c1f2b95f6ab6b60653631afd/run/force_sync.sh
Fri  6 Mar 21:55:50 GMT 2020 : Downloaded /root/bin/force_sync.sh ...
Fri  6 Mar 21:55:50 GMT 2020 : curl -o /tmp/create-backingfiles-partition.sh https://raw.githubusercontent.com/marcone/teslausb/f6e0bead51b652b2c1f2b95f6ab6b60653631afd/setup/pi/create-backingfiles-partition.sh
Fri  6 Mar 21:55:51 GMT 2020 : Downloaded /tmp/create-backingfiles-partition.sh ...
Fri  6 Mar 21:55:51 GMT 2020 : curl -o /tmp/create-backingfiles.sh https://raw.githubusercontent.com/marcone/teslausb/f6e0bead51b652b2c1f2b95f6ab6b60653631afd/setup/pi/create-backingfiles.sh
Fri  6 Mar 21:55:51 GMT 2020 : Downloaded /tmp/create-backingfiles.sh ...
Fri  6 Mar 21:55:51 GMT 2020 : curl -o /tmp/make-root-fs-readonly.sh https://raw.githubusercontent.com/marcone/teslausb/f6e0bead51b652b2c1f2b95f6ab6b60653631afd/setup/pi/make-root-fs-readonly.sh
Fri  6 Mar 21:55:52 GMT 2020 : Downloaded /tmp/make-root-fs-readonly.sh ...
Fri  6 Mar 21:55:52 GMT 2020 : curl -o /tmp/configure.sh https://raw.githubusercontent.com/marcone/teslausb/f6e0bead51b652b2c1f2b95f6ab6b60653631afd/setup/pi/configure.sh
Fri  6 Mar 21:55:53 GMT 2020 : Downloaded /tmp/configure.sh ...
Fri  6 Mar 21:55:53 GMT 2020 : Fixing the modules-load parameter in /boot/cmdline.txt...
Fri  6 Mar 21:55:53 GMT 2020 : Fixed cmdline.txt.
Fri  6 Mar 21:55:58 GMT 2020 : Checking existing partitions...
Fri  6 Mar 21:55:58 GMT 2020 : Modifying partition table for backing files partition...
[... cut ...]
Fri  6 Mar 21:58:05 GMT 2020 : make-root-fs-readonly: Mounted.
Fri  6 Mar 21:58:06 GMT 2020 : make-root-fs-readonly: Moving fake-hwclock data
Fri  6 Mar 21:58:06 GMT 2020 : make-root-fs-readonly: done
Fri  6 Mar 21:58:06 GMT 2020 : Skipping package upgrade.
Fri  6 Mar 21:58:12 GMT 2020 : All done.
Sat  7 Mar 18:54:59 GMT 2020 : Detecting whether to update wpa_supplicant.conf
Sat  7 Mar 18:56:29 GMT 2020 : Detecting whether to update wpa_supplicant.conf
Sat  7 Mar 19:07:51 GMT 2020 : Detecting whether to update wpa_supplicant.conf

marcone, 2020-03-07:
That looks fine. Please see https://github.com/marcone/teslausb/issues/35

mhuk01, 2020-03-07:
I've run those commands, but how do I access the diagnostics.txt file to post here?

mhuk01, 2020-03-07:
When I run /etc/rc.local i get this...

root@teslausb:~# /etc/rc.local
My IP address is 192.168.31.244 
Detecting whether to update wpa_supplicant.conf
root@teslausb:~# already running

marcone, 2020-03-07:
> I've run those commands, but how do I access the diagnostics.txt file to post here?

Pretty much the same way you accessed the setup log: you either use `cat` and then copy/paste it, or you can use `scp` to copy it to another computer.