marcone/teslausb #447: New Track Mode

ken830, 2020-03-08:
Release notes for the new track mode:

> "You can now save a video and data of the Track Mode driving session to a plugged in USB flash drive which must contain a folder named "TeslaTrackMode" (without the quotation marks). When "Save Dashcam for Laps" is enabled, Track Mode stores a video of each lap in a driving session when using the Lap Timer. Track Mode also stores the car status and telemetry data including details about the vehicle's position, speed, acceleration, and use of accelerator which is stored as a .CSV file on the USB flash drive."

Seems we need to update teslausb to include syncing for the new TeslaTrackMode folder and the .CSV telemetry data.

marcone, 2020-03-09:
Can someone confirm whether this TeslaTrackMode folder needs to go inside of the existing TeslaCam folder, or adjacent to it in the root? Are there subfolders inside of TeslaTrackMode? Where are the .csv files stored?

raushel, 2020-03-09:
From this article: https://tesletter.com/track-mode-v2-model-3-performance/
Like this (with data):

TeslaCam
    RecentClips
    SavedClips
    SentryClips
TeslaTrackMode
    laps-2020-03-05-21_21_39.mp4
    telemetry-v1-2020-03-05-21_21_33

miles267, 2020-03-09:
> Can someone confirm whether this TeslaTrackMode folder needs to go inside of the existing TeslaCam folder, or adjacent to it in the root? Are there subfolders inside of TeslaTrackMode? Where are the .csv files stored?

I've read elsewhere it goes in the root folder.  I'll try to find a link to it.

magicalyak, 2020-03-13:
I can confirm with my model 3 performance that It goes in root as a separate folder.

joejoethejoey, 2020-03-15:
Is there a way to manually add this folder to the TeslaUSB setup?  I've tried plugging in the Pi in to my Windows 10 PC, but it doesn't detect it.  Plugging in the MicroSD card directly only shows the BOOT partition.  All the other partitions need to be formated.

marcone, 2020-03-15:
> I've tried plugging in the Pi in to my Windows 10 PC, but it doesn't detect it.

Did you wait for the Pi boot up? Did you use the same cable you use with the car?
The Pi just acts likes a USB drive (once booted up), and it doesn't make a difference whether it's connected to a car or a PC, it will show up as a drive either way.

joejoethejoey, 2020-03-15:
Thanks for the quick Marcone!  I think my first flash of the image didn't work properly.  I've just tried again from scratch and used a micro usb cable which came with an old mobile which definitely has data and power and the PC picked it up straight after the Pi rebooted.  Cheers!!!

joejoethejoey, 2020-03-28:
I've noticed that the TeslaUSB is powered on continuously on my Model 3.  I have Sentry mode on and I've checked the samba shares on the network throughout the day and it's always available and backing up files to the NAS every once in a while.  I've not powered on the vehicle manually during the lockdown here in UK, so I'm sure the car isn't sleeping completely.  The car loses approximately 20 miles of charge throughout the day.  According to TezLab app, I've lost 128.4 miles of range while idle for 6days 5hrs.  Is this the norm now with the latest update with Sentry mode on continuously?  Cheers.

marcone, 2020-03-29:
If Sentry is on, USB is powered, therefore the Pi will be running. This has always been the case.
Note that most of the charge losses will be just from the fact that Sentry is on, not from using a Pi. You would have pretty much the same losses if you used a plain USB drive or no drive at all (Pi0 uses about half a Watt, or 0.012 kWh per day)

It should not be archiving throughout the day though, unless it loses power and restarts, or loses and regains wifi connectivity.

joejoethejoey, 2020-03-30:
Thx Marcone,

Yes, I didn't mean to imply TeslaUSB was draining the battery.  I'm guessing there's no avoiding it if we want the feature to record as and when necessary.  Is there a way to manually action an archive remotely?  What is the current scheduling as default?

closed 2020-04-19 by marcone