marcone/teslausb #448: TeslaUSB Drive Full - Stuck?

jun3280net, 2020-03-09:
Hi there,

Teslausb has been working great for me for months.  However, recently my server shared drive that is used to store the Teslausb videos filled up.  As a result Teslausb started accumulating videos (roughly 220 GB worth).  I received the Tesla USB drive is full message, so I took the raspberry pi and connected it to my computer to delete the video content through Windows.  The raspberry pi mounts properly in Windows and I can see the RecentClips, SavedClips, and SentryClips folder, but they are all empty.

[image]

I ended up SSH into the raspberry pi and ran "df -h" and discovered all the videos were still in the /backingfiles directory.  Any idea why this folder isn't synching with my server shared drive (I've cleared up space).  Conversely, any idea why this folder isn't exposed when connected to my PC in Windows - as it would make viewing and deleting Tesla video clips much easier.  I end up trying to  delete the folders via SSH - but ran into a "read-only" error message.  Any suggestions on how to easily recover disk space on the TeslaUSB would be greatly appreciated.

[image]

[image]

[image]

[image]

Sorry in advance if any of this has been answered before - but I couldn't find anything similar searching through the Issues.

Thanks,

J

marcone, 2020-03-09:
The snapshots are managed by teslausb and you don't normally need to manually manage them.

In your case it seems like something may have gone wrong since it appears to have stopped deleting snapshots. You might want to check /mutable/archiveloop.log for errors.
I'm guessing you have a 256GB sd card, and allocated most of that to the CAM drive. For a use case like that, I would expect at most one snapshot to be kept.

I would recommend starting over and specifying a far smaller camsize.

If you want to try and fix your current setup, you can certainly delete the snapshots, you just need to make sure to unmount the snapshots first, then delete the snapshots, not the files within the mounted snapshots. In fact if you were to remount the snapshots read-write and then delete the files within, you'd probably end up using *more* space, not less.

closed 2020-03-09 by marcone

jun3280net, 2020-03-09:
Thanks for the quick response Marcone.  Can you share the command line to
unmount the snapshots folder to delete the folder that would be super
helpful.  I tried a few different commands - (even one using RSYNC) without
any success.  Otherwise, I may need to restart from scratch as you've
suggested.  You are correct I have a 256GB card.  I figured it would be
useful to have something that size in the event the car was parked for an
extended period of time (eg vacation).  What would be the max CAM size you
would recommend?

Many thanks,

J

marcone, 2020-03-09:
> Thanks for the quick response Marcone. Can you share the command line to unmount the snapshots folder to delete the folder that would be super helpful. I tried a few different commands - (even one using RSYNC) without any success.

something like this might do it (untested, you may need to tweak/correct)
```
sudo -i
cd /backingfiles/snapshots
for snap in snap-*
do
  umount $snap/mnt
  rm -rf $nap
done
```

> What would be the max CAM size you would recommend?

It depends on your use case, but I find 24GB to be plenty.

jun3280net, 2020-03-10:
> something like this might do it (untested, you may need to tweak/correct)
>
> ```
> sudo -i
> cd /backingfiles/snapshots
> for snap in snap-*
> do
>   umount $snap/mnt
>   rm -rf $nap
> done
> ```
>
>
> It depends on your [...]

Thanks @marcone!
Deletion worked perfectly using your command.

for snap in snap-*; do umount $snap/mnt; rm -rf $snap; done

The one caveat is...I think the command works faster than it deletes on the raspberry pi, so I would need to use "CTRL + C" to kill the command when it said the "target is busy" when trying to unmount the directory.  Then I would just re-run the command you supplied.

jun3280net, 2020-03-11:
While the aforementioned commands help to clear up old snapshot records.  At first, the drive would register with the car and start recording.  However, over the course of a few hours I would get the Drive Full Error Message from Tesla.  I would plug the device back into the computer, and the drive would look empty.  SSH into the device to observe the snapshots, and there are some files - but not nearly enough to fill the drive to capacity.  In any case - I've taken @marcone's suggestion and and trying to reinstall from scratch.  Hopefully everything goes smoothly.