marcone/teslausb #449: exFAT support

oaquique, 2020-03-10:
Beginning with software update 2020.8.x the exFAT file system is supported for Dashcam/Sentry storage.

Are there plans to support storage formatted with exFAT with TeslaUSB?

exFAT will create smaller clusters, making write/read faster than the huge cluster created by FAT32 in large partitions. It will also make a more efficient use of space available by reducing wasted unallocated space (due to large clusters).

marcone, 2020-03-10:
According to [Microsoft](https://support.microsoft.com/en-us/help/140365/default-cluster-size-for-ntfs-fat-and-exfat) the default cluster size for a 64GB partition is 32KB for FAT32 and 128KB for exFAT, so not sure why you're saying exFAT will have smaller clusters.

Even if they were smaller, the files that the car writes are about 30MB on average, so if half a cluster is wasted on average per file, that's 0.1% overhead, i.e. negligible.

I'm not not opposed to adding support for exFAT (and [ext4](https://github.com/marcone/teslausb/issues/420)), but it's low priority for me. Feel free to submit a PR.

oaquique, 2020-03-10:
The article only includes information for FAT32 in WinNT 3.1. FAT32 for partitions bigger than 4GB has not been supported since WinNT4.0 and certainly not supported in any version of Windows.

If you use a tool like GUIformat to format a 64GB microSD card as FAT32, the suggested allocation unit somes as 32768KB (32MB) in order for FAT32 to work in such big partition. The smallest you could get (if forced) is 512KB per allocation unit.

This is considerable bigger (at least 5x more if alloc unit=512KB than exFAT).

I'll submit a PR so we could have more efficient storage available through your fine solution (I really love TeslaUSB!)

marcone, 2020-03-10:
> The article only includes information for FAT32 in WinNT 3.1. FAT32 for partitions bigger than 4GB has not been supported since WinNT4.0 and certainly not supported in any version of Windows.

Not sure where you're getting that from. The table for FAT32 in the article I linked says the cluster size for a FAT32 volume between 32GB and 2TB is 32KB. Whether or not any particular version of Windows supports formatting or accessing a particular volume size does not change the specification of the file system.

> If you use a tool like GUIformat to format a 64GB microSD card as FAT32, the suggested allocation unit somes as 32768KB (32MB) in order for FAT32 to work in such big partition. The smallest you could get (if forced) is 512KB per allocation unit.

I'm not familiar with GUIformat, but that sounds like GUIformat is either displaying things wrong, or you're misreading it, because  a 32MB cluster size would mean a total of only 2048 clusters (64GB divided by 32MB is 2048), and that's clearly bogus.

oaquique, 2020-03-10:
Just give it a try... copying files to a single FAT32 64GB partition shows much lower performance (both for read and write) than a 4GB FAT32 partition with default allocation unit.

Been there, seen that.

oaquique, 2020-03-11:
The article has a table called "Default cluster sizes for FAT32". and 4 columns: WinNT 3.51; WInNT 4.0; Win2000->Win7

The info you are referring to for 32GB to 2TB lists 32KB on the WinNT3.51 column and shows "Unsupported" for all the other columns.

Microsoft does NOT support FAT32 in partitions or volumes greater than 4GB. If you try to format a partition bigger than 4GB using any of the tools that come in Windows, you get "unsupported". I tried this both with diskpart.exe and formatting through the GUI (through the GUI only NTFS and exFAT are available as options).

Therefore, it makes more sense to use a file system that is widely supported and adopted by the SD card industry (the SD Association SD formatter will only format this 64GB microSD card in exFAT) than using FAT32.

Thanks for listening. I won't continue discussing this.

marcone, 2020-03-11:
> The article has a table called "Default cluster sizes for FAT32". and 4 columns: WinNT 3.51; WInNT 4.0; Win2000->Win7
>
> The info you are referring to for 32GB to 2TB lists 32KB on the WinNT3.51 [...]

And yet, my teslausb has a 24GB FAT32 "CAM" drive, and Windows reads and writes it just fine.
 
> If you try to format a partition bigger than 4GB using any of the tools that come in Windows, you get "unsupported". I tried this both with diskpart.exe and formatting through the GUI (through the GUI only NTFS and exFAT are available as options).

You're confusing "able to format" with "able to read and write"

> Therefore, it makes more sense to use a file system that is widely supported and adopted by the SD card industry (the SD Association SD formatter will only format this 64GB microSD card in exFAT) than using FAT32.

Like I said: I'm not opposed to supporting exFAT, it's just a low priority for me because it does not offer any real advantage for teslausb's use case. It doesn't save space, and doesn't improve speed (the bottleneck is the USB bus and drive emulation, not the filesystem)

oaquique, 2020-03-11:
Yes, you can read/write just fine as Windows allows that for backwards compatibility. However, Microsoft does not support formatting partitions larger than 4GB with FAT32.

Now, you are confusing “not supported” with “not able to read/write”.

marcone, 2020-03-11:
> Yes, you can read/write just fine as Windows allows that for backwards compatibility. However, Microsoft does not support formatting partitions larger than 4GB with FAT32.

So? Teslausb handles the formatting of the drives, so it doesn't matter that the standard Windows tools don't allow formatting larger than 4GB FAT32.

> Now, you are confusing “not supported” with “not able to read/write”. 

You're the one that said that "Microsoft does NOT support FAT32 in partitions or volumes greater than 4GB." If by that you meant "does not support formatting, but can still read and write", then by that definition Tesla does not support exFAT either 😊

In any case, regardless of whatever merits exFAT may have, some of the things you said above about FAT32 are just plain wrong. The cluster size is not 32MB, it's 32KB (the number that GUIformat/FAT32format displays is bytes, not kilobytes), so you're not going to get noticable space saving from using exFAT for teslausb's use case (which is writing large files), and as I already mentioned the speed bottleneck is not the filesystem but the USB bus and drive emulation.

Again, happy to add support for exFAT, it's just not a high priority for me.

I'm going to close this and mark it as a duplicate of https://github.com/marcone/teslausb/issues/420, and a note there to add support for both ext4 and exFAT.

closed 2020-03-11 by marcone

oaquique, 2020-03-11:
I said Microsoft doesn't support FAT32 in partitions larger than 4GB. It means, Windows won't provide the means to format such partitions and whatever happens with larger partitions, you are on your own.

The extrapolation and assumption you make about exFAT is simply wrong.

Do whatever you want, and learn how to be humble and to treat people with respect.

Closed #449<https://eur05.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmarcone%2Fteslausb%2Fissues%2F449&data=02%7C01%7C%7Ce074824997384cc50a3408d7c55b0f56%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637194867191258107&sdata=S1gIDCMSfnslgld6zMaPTQXmIx7TFYP7EU16Deg%2F1vY%3D&reserved=0>.

marcone, 2020-03-11:
> I said Microsoft doesn't support FAT32 in partitions larger than 4GB. It means, Windows won't provide the means to format such partitions and whatever happens with larger partitions, you are on your [...]

I made no assumptions and extrapolations about exFAT whatsoever, and if me pointing out that you were spreading incorrect information about FAT32 is disrespectful in your mind, then perhaps it is you that needs to learn how to be humble.

oaquique, 2020-03-11:
There are many ways to convey an idea. You can do it kindly or you can go the self-entitled "I'm better than you" way. You always choose the second.

The disrespect is not in saying I could be ("spreading" according to you) mistaken about FAT32 but in how to say it. Your communication style is entitled and arrogant.

You could say "I think you may be confused because this or that" instead of "you are plainly wrong."

I have no problem in recognizing when I make a mistake, and it clear -based on this thread- you don't.

This is not the first time I have seen this behavior and tone when interacting with you, so it is not driven by the conversation about FAT32.

Good luck!