marcone/teslausb #450: TeslaUSB Drive Full - Stuck Still?

jun3280net, 2020-03-12:
Continuing from my last issue #448 - I followed @marcone 's instruction and reinstalled from scratch.
1.  I wiped the micro SD Card using SD Card Formatter
2.  I used Etcher to install the teslausb-buster-20190922.zip (because I tried teslausb-20200122.zip and it hung.  Also the buster image was working for me).
3.  Copied the teslausb_setup_variables (same configuration as what I had working) to the boot drive
4.  Plugged the micro SD Card into the Raspberry Pi and let it complete the installation.
5.  SSH to confirm the installation completed properly
6.  Plugged the Raspberry Pi into Tesla Model S
7.  Confirmed it was working and drove into work.
8.  After work (9.5 hrs later) returned to the car - and noticed the camera was no longer recording.  Tesla said "Dashcam unable to write - usb drive is full".  This is a 256GB card with 187GB allocated for video recording.
9.  What's interesting from the SSH - it shows less than 10GB was used for the video snapshots (0-10), which is nowhere near capacity.
10.  Plugged the PI into my computer and the CAM drive shows 524MB of recordings in the RecentClips folder.

[image]

[image]

[image]

Does anybody have an idea what is going on?  Is the micro SD card corrupt?  Is this a problem because of the latest Tesla Update (2020.8.1 for Model S MCU1 / AP2.5).  

On a separate note, given that the Snapshots (0-10) still exist per SSH, is there anyway to easily expose them to Windows like storing them on a separate partition (like CAM - have an ARCHIVE.bin)?

In the meantime, I'm going to take a smaller 64GB card and retry the installation.

softcoder, 2020-03-12:
Is it a fake sd card? This kind of result could occur from a fake Chinese version:

https://www.raymond.cc/blog/test-and-detect-fake-or-counterfeit-usb-flash-drives-bought-from-ebay-with-h2testw/

https://youtu.be/J-D6tYBX8vE

jun3280net, 2020-03-12:
> Is it a fake sd card? This kind of result could occur from a fake Chinese version:
> 
> https://www.raymond.cc/blog/test-and-detect-fake-or-counterfeit-usb-flash-drives-bought-from-ebay-with-h2testw/
> 
> https://youtu.be/J-D6tYBX8vE

Thanks for the tip @softcoder .  I'll give those tools a test.  That said, I was using this card for a number of months successfully without issue.  I've moved/archived several hundred gigs of videos over that period.

jun3280net, 2020-03-13:
@softcoder did one test on the card, and it passed.  Currently, running a more intensive second test - its still going but no errors.  

On a separate note, I used a 64GB card that I initially did testing with for TeslaUSB - did a fresh install.  Everything installed smoothly, even my 700MB library of music synced properly.  The moment I plugged the drive into the car - it registered as full.  

Now I'm beginning to wonder if it's related to my car configuration/software.  Will try a few more tests.

jun3280net, 2020-03-13:
Reinstalled the 64GB drive, and pretty much the same result.  I started from scratch which includes the server.  Reduced the amount of music to be transferred, which seemed to give the card a bit of room to write to.  That said - it filled up fast in snapshots (less than 150MB) - and Tesla complained the drive was full.  I've tried plugging/replugging the TeslaUSB - and the car sees the drive as full.

[image]

[image]

marcone, 2020-03-14:
Have you tried just using a regular USB drive? Because from the Pi's point of view there is no difference between being connected to a Tesla, a PC, or any other computer, and so the fact that the drive shows up correctly (empty) in Windows while the car thinks it's full might indicate a problem with the car.

jun3280net, 2020-03-14:
> Have you tried just using a regular USB drive? Because from the Pi's point of view there is no difference between being connected to a Tesla, a PC, or any other computer, and so the fact that the [...]

@marcone yup that's the next step.  Last test I tried was 48% Cam / 2% Music - and same result.  The only other thing I haven't tried is turning off the snapshots.  

That said - i'm just going to wipe my 64GB card, and use it as standard storage to see how long it goes before the system complains that the drive is full.  Thanks for the suggestion.  Will update tonight after my hockey game.

marcone, 2020-03-14:
you should also check `dmesg` on the Pi to see if there are any I/O errors on the card

jun3280net, 2020-03-14:
@marcone - thanks for the tips.  Using just a normal USB drive - same result.  Pretty sure the issue is related to the updates from Tesla - going to see if other people experiencing the same thing.  In the meantime, the dmesg is as follows : 

pi@teslausb-mdls:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.93+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1290 Fri Jan 10 16:34:37 GMT 2020
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] On node 0 totalpages: 114688
[    0.000000]   Normal zone: 1008 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x494 with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:7D:93:52 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=27cf9d8e-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait modules-load=dwc2 quiet fastboot noswap ro
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 434164K/458752K available (7142K kernel code, 656K rwdata, 2128K rodata, 468K init, 797K bss, 16396K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
                   lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (7144 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   ( 468 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 657 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 798 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 25651 entries in 76 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000030] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000076] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000198] bcm2835: system timer (irq = 27)
[    0.000616] Console: colour dummy device 80x30
[    0.000644] console [tty1] enabled
[    0.000705] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060347] pid_max: default: 32768 minimum: 301
[    0.060908] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060930] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062372] CPU: Testing write buffer coherency: ok
[    0.063680] Setting up static identity map for 0x8200 - 0x8238
[    0.065091] devtmpfs: initialized
[    0.076417] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.076822] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.076853] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.078195] pinctrl core: initialized pinctrl subsystem
[    0.079786] NET: Registered protocol family 16
[    0.083263] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.091034] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.091051] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.091198] Serial: AMBA PL011 UART driver
[    0.094519] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.140002] bcm2835-dma 20007000.dma: DMA legacy API manager at (ptrval), dmachans=0x1
[    0.142706] SCSI subsystem initialized
[    0.142982] usbcore: registered new interface driver usbfs
[    0.143089] usbcore: registered new interface driver hub
[    0.143317] usbcore: registered new device driver usb
[    0.160982] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-01-10 17:01, variant start
[    0.171001] raspberrypi-firmware soc:firmware: Firmware hash is f3e39fed4763ce4efbffabb2d4787aaacddeda0f
[    0.183699] clocksource: Switched to clocksource timer
[    0.291292] VFS: Disk quotas dquot_6.6.0
[    0.291424] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    0.291751] FS-Cache: Loaded
[    0.292102] CacheFiles: Loaded
[    0.311779] NET: Registered protocol family 2
[... 181 lines cut ...]
[   29.946120] CIFS VFS: Error connecting to socket. Aborting operation.
[   29.946150] CIFS VFS: cifs_mount failed w/return code = -101
[   29.954074] CIFS VFS: Error connecting to socket. Aborting operation.
[   29.954109] CIFS VFS: cifs_mount failed w/return code = -101
[   30.986496]  loop0: p1
[   32.430016]  loop1: p1
[... cut ...]
[  115.545891] LUN: removable file: /backingfiles/music_disk.bin
[  115.545991] LUN: removable file: /backingfiles/cam_disk.bin
[  115.546001] Number of LUNs=2
[  115.550542] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[  115.550559] g_mass_storage gadget: g_mass_storage ready
[  115.550574] dwc2 20980000.usb: bound driver g_mass_storage
[  115.950705] dwc2 20980000.usb: new device is high-speed
[  116.026961] dwc2 20980000.usb: new address 13
[  116.064460] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage

sparky742, 2020-03-14:
I setup mine when the car was on 2020.4.x   worked great.  Since then, I've been through 2020.8 for a day, then 2020.12 and noticed that it failed last time I had a sentry event with "reformat the drive in a supported format" and "USB drive too slow"...so it may be the Tesla update that did it. I believe since these two updates, exFAT is now supported.

jun3280net, 2020-03-15:
It took a while to find - but it appears this issue is likely related to the software update : 

[https://teslamotorsclub.com/tmc/threads/dash-cam-error-says-drive-is-full-but-its-nearly-empty.188108/#post-4564358](https://teslamotorsclub.com/tmc/threads/dash-cam-error-says-drive-is-full-but-its-nearly-empty.188108/#post-4564358)

drlauridsen, 2020-04-02:
Same issue here, both with raspberry pi and standard USB drive.. Posted the issue on Twitter, maybe "like" if you have the issue..
https://twitter.com/JohnnyLauridse4/status/1245619926456401920?s=19

sparky742, 2020-04-02:
yeah, I was guessing the exFAT change caused this.  Further to your tweet - my M3 is totally working fine with my FAT USB stick back in place, just not the rPi + TeslaUSB.

drlauridsen, 2020-04-02:
> yeah, I was guessing the exFAT change caused this. Further to your tweet - my M3 is totally working fine with my FAT USB stick back in place, just not the rPi + TeslaUSB.

Strange. My FAT USB stick has the same error after some time..

jun3280net, 2020-04-14:
> > yeah, I was guessing the exFAT change caused this. Further to your tweet - my M3 is totally working fine with my FAT USB stick back in place, just not the rPi + TeslaUSB.
> 
> Strange. My FAT USB stick has the same error after some time..

Just an update - looks like this issue has been addressed with 2020.12.5 update.  Will close this off once I can confirm things are good over the next couple days.

jun3280net, 2020-04-16:
Not out of the woods yet - can anyone confirm whether their Music partition works?  If so, is the Music partition Fat32?  My videos are being recorded and synced, but my Music partition is not recognized in the car (Model S MCU1/AP2.5 2020.12.5).

jun3280net, 2020-04-17:
2020.12.5 Fixes the dash cam recording issue which affected Model S/X (MCU1/AP2.5).  However it introduces the issue where the Music partition / USB Drive is no longer recognized.

closed 2020-04-17 by jun3280net

jun3280net, 2020-04-26:
> 2020.12.5 Fixes the dash cam recording issue which affected Model S/X (MCU1/AP2.5). However it introduces the issue where the Music partition / USB Drive is no longer recognized.
> 
> https://teslamotorsclub.com/tmc/threads/2020-12-5.190487/page-6

The music partition is now recognized too with update 2020.12.10