marcone/teslausb #452: Setup Problem - Stuck

mynameisgv, 2020-03-15:
Hello 
I tried the one step setup instructions. My wifi details are right. 
It creates a file called WIFI_ENABLED and setup started and is stuck  after that. 
In logs i notice that the date is wrong. I expected PST. Sat 14. Time looks right.

$ cat teslausb-headless-setup.log
Sat 15 Jun 22:27:33 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sat 15 Jun 22:27:33 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Sat 15 Jun 22:27:33 BST 2019 : Rebooting...
Sat 15 Jun 22:28:19 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sat 15 Jun 22:28:19 BST 2019 : Grabbing main setup file.
Sat 15 Jun 22:28:19 BST 2019 : get_script failed, retrying
Sat 15 Jun 22:28:23 BST 2019 : get_script failed, retrying
Sat 15 Jun 22:28:26 BST 2019 : get_script failed, retrying
Sat 15 Jun 22:28:29 BST 2019 : get_script failed, retrying

I have a basic configuration.
set ARCHIVE_SYSTEM to none and 
set the SSID and WIFIPASS to the right one. 
my wifi is 2.4ghz. 

Any pointers will help
Thanks

marcone, 2020-03-15:
The "get_script failed" error is likely because the date on the Pi is wrong. Time and date are normally set by NTP.

nebukadnezar, 2020-03-16:
same problem here, but my date is Jan 20 2020... any suggestions? Tried it on two different Wifi's, and it doesn't ever seem to even register on the wifi. AP ever get a DHCP request from the Pi.

marcone, 2020-03-16:
If it's not connecting to wifi then obviously NTP isn't going work.
You'll need to solve your wifi problem first.

nebukadnezar, 2020-03-16:
that's the problem, it won't connect to any of my wifis. I now tried the previous release 2019-09-22 and that one seems to work. Odd.

closed 2020-03-26 by marcone