marcone/teslausb #453: Working without an archive - just copying all clips all the time to an external disk?

nebukadnezar, 2020-03-18:
My use case appears to be slightly different from everyone else's here: I simply want teslausb to continuously copy/rsync all clips (mostly RecentClips) to an external disk. I don't have wifi in my parking so that defined "archive stop" never happens, I need to run Tesla software updates off a portable 4G hotspot and that's not suitable for archiving anyway due to the data volumes involved. I live in a land where speed cameras are rampant, and the authorities are quite haphazard with their infringement notices. I recently for example drove through a supposedly temporarily lowered speed limit and got blitzed without me noticing, but the 1h clip buffer of course was long overwritten by the time I received the fine in the mail (1-2 weeks later) so I couldn't prove the limit wasn't actually lowered on the signs. So all I'd really want to do is have the RecentClips archived to an external disk before the 1h buffer is up. I've tried to work around this by adding my own rsync code that rsyncs the contents of /backingfiles/snapshots/snap-000000/mnt/ to the external disk, but that doesn't seem to work for reasons I don't quite understand (but likely because the archive state never goes to "archive available" = true. Any suggestions on how I could implement this?

marcone, 2020-03-18:
Sounds like you should just use a Pi4 with a large external drive.  You can configure teslausb to use that for storage directly, no need to rsync to it from one drive to the other.
Then when you need to, access the stored footage via Samba.

nebukadnezar, 2020-03-19:
I'm not having much luck. What should I select for the ARCHIVE_SYSTEM option in the config file for this? I've tried 
`export ARCHIVE_SYSTEM=none`
 and 
`export usb_drive=/dev/sda1`

In the past 5 attempts (using two different SD cards) I'm not getting any success at all, the pi won't connect to the wifi and therefore can't git the setup script.

On the 2019 image, it makes it past that hurdle but then gets stuck here (file from the future problem?).

Sun 22 Sep 22:54:54 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sun 22 Sep 22:54:54 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Sun 22 Sep 22:54:54 BST 2019 : Rebooting...
Sun 22 Sep 22:54:54 BST 2019 : Grabbing main setup file.
Sun 22 Sep 22:55:08 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sun 22 Sep 22:55:08 BST 2019 : Grabbing main setup file.
Sun 22 Sep 22:55:08 BST 2019 : get_script failed, retrying
Sun 22 Sep 22:55:12 BST 2019 : Starting setup.
Sun 22 Sep 22:55:12 BST 2019 : adding country code to wpa_supplicant.conf
Sun 22 Sep 22:55:12 BST 2019 : increased root partition size
Sun 22 Sep 22:55:12 BST 2019 : rebooting for changes to take effect
Sun 22 Sep 22:55:29 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sun 22 Sep 22:55:29 BST 2019 : Starting setup.
Sun 22 Sep 22:55:29 BST 2019 : increasing root filesystem size to match partition size
Sun 22 Sep 22:55:30 BST 2019 : reading config from /root/teslausb_setup_variables.conf
Sun 22 Sep 22:55:30 BST 2019 : curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev
Sun 22 Sep 22:55:50 BST 2019 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Sun 22 Sep 22:56:13 BST 2019 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Sun 22 Sep 22:56:37 BST 2019 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying

marcone, 2020-03-19:
> I'm not having much luck. What should I select for the ARCHIVE_SYSTEM option in the config file for this? I've tried
> `export ARCHIVE_SYSTEM=none`
> and
> `export usb_drive=/dev/sda1`

That should be "/dev/sda", the device name of the disk itself.

> In the past 5 attempts (using two different SD cards) I'm not getting any success at all, the pi won't connect to the wifi and therefore can't git the setup script.
> 
> On the 2019 image, it makes it past that hurdle but then gets stuck here (file from the future problem?).

Not sure why it doesn't connect to wifi for you with the latest image, but having the time set incorrectly will generally cause setup to fail. Time should be set by NTP once connected to wifi.

nebukadnezar, 2020-03-19:
Is there a way to turn out verbose logging so it will preserve system level diagnostic output in the headless setup log file?

marcone, 2020-03-19:
You can ssh in to the Pi and run `/root/bin/setup-teslausb` manually.

nebukadnezar, 2020-03-19:
headless setup finished this time, but I used the ethernet connection instead of wifi. Not sure what's wrong with the wifi part of the setup, but it would only connect every 5th time or so, hence the occasional success. On the setup side it appears to have correctly configured the external disk for the snapshots using /dev/sda as you indicated above.

I'm now in my car and can connect via smb to the pi and see any clips I'm saving by tapping on the camera icon on the MCU.

Will it save all of the RecentClips as well? I don't have to time to go drive for a few hours at the moment to verify this works. Or do I need to customise anything to make it save all RecentClips (and not just the SentryClips and SavedClips)?

marcone, 2020-03-19:
It snapshots the entire drive, including RecentClips

nebukadnezar, 2020-03-20:
Ok, that seems to work. Sort of. After about an hour of driving, the drive becomes unavailable (i.e. corrupted). And it stays like that until I reboot the pi. Any recommendations on how the pi can fix this on its own while running? Cronjob to reboot once an hour?

marcone, 2020-03-20:
Drives shouldn't just get corrupted like that, and rebooting isn't a solution, it's just masking the problem.
I would first look in to making sure that your Pi and drive have a stable power supply.
(I assume you're using an SSD or other flash drive and not a spinning disk? Because that's probably not good to have in a car)

nebukadnezar, 2020-03-20:
I'm using a Samsung T5 SSD and powering the combo off the front USB connector in the Tesla. I haven't noticed any power related issues such as random reboots - it stays up and running for as long as the port is powered. So from what I read teslausb does a self-repair when the drive the car writes to becomes corrupted. How does it know the drive has gone bad?

marcone, 2020-03-20:
It checks the file system every time it boots, and before it starts archiving.
Filesystem corruption should really only happen due to hard power off, e.g. when the car powers off the USB ports, which shouldn't happen during driving.

nebukadnezar, 2020-03-20:
How is the archive function triggered? I'm about to embark on a 10h drive home today and could be inserting some test code. But I don't really understand what's triggered and when. I'm not in the car right now but from memory there's an archive_available script that is called by the archive_loop which in my case always returns false (false is the only line in there in fact). Yet somehow it does archive data to the external disk.

marcone, 2020-03-20:
That's not archiving, that's snapshotting.
The drive that the car sees is actually a disk image in a file located on the external drive (/backingfiles/cam_disk.bin), and teslausb takes hourly snapshots of that disk image file.

Archiving is when teslausb copies the individual recordings to a file share or cloud service.

nebukadnezar, 2020-03-20:
That clears up my fundamental misunderstanding of how this works, thanks! So would I expect to see a whole series of (and again, not in the car, so from memory): snap0000, snap0001 etc. snapshot files on the external disk? It looked like the samba server was showing me files from all of yesterday (minus the times it had failed) - so does samba present a consolidated list of all files in all snapshots? I have never seen a snapshot file mounted other than the index 00000 one.

marcone, 2020-03-20:
Yes, the snapshots are numbered in sequence. When the drive starts to become full it will start deleting older snapshots, so the oldest snapshot is not necessarily number 0.
As you say, the samba server presents a consolidated list of all the files in all snapshots. There can be dozens or even hundreds of snapshots, but if you set "camsize" to 100%, there will only ever be at most 1 snapshot.

nebukadnezar, 2020-03-23:
I've now set this up now so that I'm using the Pi4 with a 256GB SD card, and using 100% for the camfiles. The dashcam icon on the MCU display now stays on and indicates recording all the time, and it doesn't seem to get corrupted like it used to when using the external disk. But now I have a different problem: The RecentClips directory invariably only contains clips from when I parked the car, i.e. SentryClips. Not a single clip from while driving remains on the samba share, except the ones that I save manually by tapping the camera icon on the MCU, plus clips in SentryClips. The SentryClips are duplicates of what's in RecentClips, but there are more clips in RecentClips from while parked, but none from while driving. Also, samba shows the partition to be of ~6GB size. I would have expected that partition to be on order 256GB - 2GB for the system partition - 16GB for the active writing partition = 238GB in size. 

/backingfiles/cam_disk.bin however is 227GB in size, so much larger than the samba exposed partition. Also, the samba mount point is into /backingfiles/TeslaCam as per /etc/samba/smbd.conf, which is the partition the car is writing to, so I suspect what you said earlier that samba exposes the merged archive isn't really the case.

Looking at /backingfiles/snapshots/snap-000000/mnt/TeslaCam (the large snapshot partition) then has a completely empty RecentClips directory, and only the SentryClips and SavedClips have data in them.

I'm confused.

marcone, 2020-03-23:
> I've now set this up now so that I'm using the Pi4 with a 256GB SD card, and using 100% for the camfiles. The dashcam icon on the MCU display now stays on and indicates recording all the time, and it doesn't seem to get corrupted like it used to when using the external disk. But now I have a different problem:
> 
> On the samba mount, the RecentClips directory invariably only contains clips from when I parked the car, i.e. SentryClips. Not a single clip from while driving remains on the samba share, except the ones that I save manually by tapping the camera icon on the MCU, plus clips in SentryClips. The SentryClips are duplicates of what's in RecentClips, but there are more clips in RecentClips from while parked, but none from while driving. Also, samba shows the partition to be of ~6GB size. I would have expected that partition to be on order 256GB - 2GB for the system partition - 16GB for the active writing partition = 238GB in size.

What do you mean by "16GB for the active writing partition"? You set `export camsize=100%`, so as you saw your cam_disk.bin will be 227GB (the entire size of the card, minus some for overhead).
With that size, there is essentially no room for snapshots, and so what you're seeing via Samba is just a temporary snapshot which will be deleted when the next one is taken.

> /backingfiles/cam_disk.bin however is 227GB in size, so much larger than the samba exposed partition. Also, the samba mount point is into /backingfiles/TeslaCam as per /etc/samba/smbd.conf, which is [...]

The car does not write to directly the /backingfiles partition. As I said in a previous comment, the drive that the car sees is actually a disk image file, /backingfiles/cam_disk.bin. It is the size of that disk image file that you specify in the config file. I suggest you set it to `export camsize=16G`

> Looking at /backingfiles/snapshots/snap-000000/mnt/TeslaCam (the large snapshot partition) then has a completely empty RecentClips directory, and only the SentryClips and SavedClips have data in them.

That's because the car constantly deletes RecentClips that are older than one hour, and you haven't given teslausb enough free space to preserve those recordings by making snapshots.

nebukadnezar, 2020-03-23:
thanks for your patience... I misread your comment above, teslausb takes snapshots OF cam_disk.bin as you reiterate, not FROM cam_disk.bin. I'll reconfigure and try again.

nebukadnezar, 2020-03-24:
Now we seem to have success! Thanks for your assistance with getting this working!

closed 2020-03-26 by marcone