marcone/teslausb #454: So I have tried multiple times to get my car to record clips to teslausb... no joy

louismkahn, 2020-03-23:
So I have tried both v1.3 and 2.0 Beta 2... in both cases got the Pi Zero W fully setup, shows up on Wifi, etc... nice job on the auto-setup scripts... but... after it gets all the way to the 2 blinks a second stage which should mean it is done and ready to plug into my car, I plug it in, wait for it to boot up and it gets back to the 2 blinks about every second after a good 45 seconds to a min... but the car never shows the camera USB recording icon... it's like it does not recognize the Pi... if I plus the USB hard disk back in things work fine.... any clues for next steps for troubleshooting getting the car to actually recognize this setup? 
Thank you

marcone, 2020-03-23:
99% of the time the problem is the USB cable, i.e. using a charge-only cable.
So try different cables until you find one that works.

louismkahn, 2020-03-23:
That was it. Thank you.

closed 2020-03-23 by marcone