marcone/teslausb #455: No CAM folder after succesful setup

Ramdg33, 2020-03-26:
I had teslausb 20200122 running on a Pi0W but I decided to replace my 32gb for a 128gb sd card to increase music folder. I can't get the system to work anymore. The setup seems to be succesful but the CAM and MUSIC folder don't show up after the reboot. I have tried other cables, sd cards and other images. The setup logfile shows a failure when updating the index files. I have logged in to the Pi, created r/w and updated with sudo apt-get and exit. Unfortunately no changes. When I tried to ssh to the Pi it showed a key error so I removed the .ssh/known_hosts file, since i only use ssh to login to the Pi. I have checked everything I'm aware of during the last week but now I need your help.

marcone, 2020-03-26:
The setup log looks fine. The diagnostics are consistent with using a charge-only cable or using the wrong port on the Pi.
Most of the time, the problem is the cable.

Ramdg33, 2020-03-26:
Thx! I used the usb port so that should be good. I'll try another cable without the usb hub and let you know how it went. Do you have any idea why the index update fails?

marcone, 2020-03-26:
it failed temporarily (otherwise it would have kept retrying) because the correct time wasn't set yet.

Ramdg33, 2020-03-26:
I've read this omewhere that the time is set later. Is there a way to fix this? It's annoying me. Can I move up the time in the configfile?

Ramdg33, 2020-03-26:
I just did a new setup with another cable and all is good now, thx!! Just interested to know if I can fix this index issue by moving the time up in the config file, for instance right after configuring the hostname?

marcone, 2020-03-26:
I'm not sure I understand what you want to move. The time is not set by the config file.
In any case, there is really no need to do anything. Everything is working as expected and the package index was updated.

closed 2020-03-26 by marcone

Ramdg33, 2020-03-27:
Ok I wasn't aware time is not set by config file. I'll leave it since everything is working now. Thx for your support and making teslausb possible!!