marcone/teslausb #458: Installing TeslaUSB

alphatech69, 2020-03-29:
Marconi, fantastic job with this.  I have a quick question, how do I trigger file sharings (Sentry Clips) from Raspberry to my NAS.  I would like to test it before I install it in my car.  Yesterday, my initial testing was successful with copying my music files from the NAS to USB but I do not see anything going from the Raspberry to NAS and although both the music and TesCam use the same NAS configuration. My understanding is that as soon as it sees the wifi and connects to it, after 20 seconds it will start moving files.  There is another way to trigger that it connected to the wifi?

Thanks

marcone, 2020-03-29:
`touch /tmp/archive_is_unreachable` will trigger an archive cycle.

closed 2020-03-29 by marcone

alphatech69, 2020-03-29:
Thank you Marcone, very much appreciated.  I'm trying to map both shares to QNAP NAS system.  During the setup, your script seems to map both drives correctly because I do not see any errors but when I do df -h I do not see it mounting to //192.168.0.20 is this normal? 

Thanks again

marcone, 2020-03-29:
See https://github.com/marcone/teslausb/wiki/FAQ#q-i-sshd-into-the-pi-but-am-not-seeing-my-network-share

alphatech69, 2020-03-29:
Then something I'm doing wrong because I see the music share sync but the TeslaCam on the NAS is empty.

marcone, 2020-03-29:
Check /mutable/archiveloop.log

alphatech69, 2020-03-29:
it is giving me permission denied.  I'm using sudo -i and making the directory readable /root/bin/remountfs_rw

marcone, 2020-03-29:
What is giving you permission denied?

alphatech69, 2020-03-29:
when I type /mutable/archiveloop.log

marcone, 2020-03-29:
That's not an executable script :)
It's just a logfile, so you'll need to run something like `cat /mutable/archiveloop.log` or open it in an editor.

alphatech69, 2020-03-29:
sorry, I have been working all day on this fun project, I enjoy the learning progress.  I re-read your FAQ and I think the problem is that I do not have any videos from Sentry, I will do a test run tomorrow morning. 
Thank you again

fscotland, 2023-03-31:
@alphatech69 , did you ever get your Telsa USB to sync with your NAS? If so, how? I am getting an error, "not keeping car awake". I am researching it now.

marcone, 2023-04-01:
> I am getting an error, "not keeping car awake".

That's not an error, it's just a message telling you that TeslaUSB will not try to keep the car awake using the Tesla API for the duration of the sync. The most likely reason for that is that you did not specify a refresh token.
(see https://github.com/marcone/teslausb/blob/main-dev/pi-gen-sources/00-teslausb-tweaks/files/teslausb_setup_variables.conf.sample#L240)