marcone/teslausb #459: Notifications are sent prior the files…

sebnaf, 2020-03-30:
The notifications should be sent after the transfer, ideally reflecting the status code of the transfer.

Thanks!

```
Mon 30 Mar 18:12:08 CEST 2020: Sending IFTTT message for moved files.
Mon 30 Mar 18:12:26 CEST 2020: Moving clips to archive...
Mon 30 Mar 18:12:26 CEST 2020: Creating output directory 'SentryClips/2020-03-30_17-09-50'
Mon 30 Mar 18:12:26 CEST 2020: Moving 'SentryClips/2020-03-30_17-09-50/2020-03-30_16-59-17-front.mp4'
Mon 30 Mar 18:12:49 CEST 2020: Moved 'SentryClips/2020-03-30_17-09-50/2020-03-30_16-59-17-front.mp4'
Mon 30 Mar 18:12:49 CEST 2020: Creating output directory 'SentryClips/2020-03-30_17-09-50'
Mon 30 Mar 18:12:49 CEST 2020: Moving 'SentryClips/2020-03-30_17-09-50/2020-03-30_16-59-17-left_repeater.mp4'
Mon 30 Mar 18:13:12 CEST 2020: Moved 'SentryClips/2020-03-30_17-09-50/2020-03-30_16-59-17-left_repeater.mp4'
Mon 30 Mar 18:13:12 CEST 2020: Creating output directory 'SentryClips/2020-03-30_17-09-50'
Mon 30 Mar 18:13:12 CEST 2020: Moving 'SentryClips/2020-03-30_17-09-50/2020-03-30_16-59-17-right_repeater.mp4'
```

DMBlakeley, 2020-03-30:
The IFTTT option is a great way to get notifications. I get one when the archive starts. When finished I get a second notification of how many files have been transferred and if any errors.

marcone, 2020-03-30:
> The notifications should be sent after the transfer, ideally reflecting the status code of the transfer.

It does. It sends a message before it starts archiving, and a message afterward.

closed 2020-03-30 by marcone