marcone/teslausb #461: The cam and music drives don't go up on my MAC's desktop

Titanium31, 2020-04-01:
Hello,

Everithing seems to be ok but the cam and music drives don't go up on my MAC's desktop. and so my model 3 doesn't see them either.
[Diagnostics.txt](https://github.com/marcone/teslausb/files/4414792/Diagnostics.txt)

The teslausb-headless-setup.log ends by All done.
The TESLAUSB_SETUP_FINISHED was created.
The led on the PI 0W flash two times.
The wifi from the pi is up
The music on my NAS seems to be copied on

marcone, 2020-04-01:
99% of the time this is caused by using a charge-only cable, and your diagnostics are consistent with that. So try different cables.

closed 2020-04-01 by marcone

Titanium31, 2020-04-01:
Hello

It's strange because it's a new installation following parameters (wifi) that I changed, I preferred to start from a clean install.

I had previously installed it on the same pi 0W and the same cable and everything worked.

I'm still going to change the cable.

Daniel

Titanium31, 2020-04-01:
I've tried 2 other cables with the same behavior...

Daniel

marcone, 2020-04-01:
Do you have access to a Windows PC? I've found that a good way to check the USB connection is to just connect the bare Pi, without the sd card, to a Windows PC. The Pi should show up as "BCM2708 Boot" in Device Manager.

Titanium31, 2020-04-01:
Yes i have a pc, i will try asap !

Thank’s

Daniel

Titanium31, 2020-04-01:
I just try and it does’nt show up in the device manager. So do you think that the pi 0 usb/power port is dead ?

Thanks again.

Daniel

marcone, 2020-04-01:
> I just try and it does’nt show up in the device manager. So do you think that the pi 0 usb/power port is dead ?

Note that the USB and power ports are different ports, and they're not interchangeable. You **must** use the port marked "USB" to connect to the computer/car.
If that's what you're doing, then either the port or the cable is bad. I'd try a few more cables, or at least try your existing cables with a different device (like a phone or a Kindle) to verify that the cables are good.

Titanium31, 2020-04-01:
but the pi zero, have two USB ports one of them is used also for power, right. I used the usb port named PWR.
Should i use the two ports one for power and the other connected to the car ?
It works before with only the PWR port !

Thank’s.

Daniel

marcone, 2020-04-01:
> but the pi zero, have two USB ports one of them is used also for power, right. I used the usb port named PWR. Should i use the two ports one for power and the other connected to the car ? It works before with only the PWR port !

No, it can't have worked with just the PWR port, because the PWR port does not have the data lines required for communication with the car or computer. It **only** provides power.

The port marked USB has both data and power lines, so it can power the Pi and communicate with the car/computer through the same port.

Titanium31, 2020-04-02:
Hello,

	I firmly believed it was the PWR port that did power and USB. So I was wrong. During my first installation I had used the USB port, and when I changed, I switched to the other port. I'm really stupid …
Everything's working properly now.

Thank you so much for your help.

Daniel

Titanium31, 2020-04-02:
Actually I was using the wrong port on the pi, the PWR and not the USB. I wrongly thought that it was the PWR port that provided the power and the USB....

Problem solved thanks to marcone.

Daniel