marcone/teslausb #462: Force upload/sync via crontab?

sebnaf, 2020-04-06:
Often the car is parked for e.g. a whole week in front of the house.
During this time the recorded sentry videos should be uploaded.

Is it safe to trigger "touch /tmp/archive_is_unreachable" via crontab e.g. every night at 3am?

Thanks!

marcone, 2020-04-06:
Depends on what you mean by "safe".
It will trigger an archive cycle, but will make the drive not visible to the car for the duration of the archive operation, i.e. you won't get recordings during that time. It also used to be that the car would sound the alarm if the drive was removed while sentry was active, but I don't know if it still does that.

sebnaf, 2020-04-06:
Thanks for the feedback.
In this case I‘ll take the wlan Access Point offline for 30 minutes.

Don’t want to risk to set the alarm off.

Thanks again!

marcone, 2020-04-06:
Not sure what you mean by that. If your access point is offline, how will the Pi do any archiving?

sebnaf, 2020-04-07:
The Pi will start archiving again when connected to WLAN. So I can "trigger" an archive process by taking the access point offline and online. That helps.

Thanks!

closed 2020-04-07 by sebnaf

marcone, 2020-04-07:
Note that taking the access point offline and back online will have the same effect as `touch /tmp/archive_is_unreachable`: they both trigger an archive cycle, and the drive is unavailable to the car for the duration of the archive cycle.