marcone/teslausb #463: Updating Instructions

pgib, 2020-04-10:
This was the process I followed to update to the latest code. If it sounds like a sound way of doing it, I'll happily submit a PR to the README to include:

1. SSH into the Pi
2. Become the root user: `sudo bash`
3. Remount the filesystems as read/write: `/root/bin/remountfs_rw`
4. Copy back your existing config: `cp /root/teslausb_setup_variables.conf /boot`
5. Remove the setup finished file: `rm /boot/TESLAUSB_SETUP_FINISHED`
6. Reboot: `reboot`

The Pi will reboot, run though the setup procedure again which will download the active scripts from Github. Note that if you set `REPO` and/or `BRANCH` in your `teslausb_setup_variables.conf` to be different, it will use that as the source. After running through the setup procedure, the Pi will be reboot once again.

marcone, 2020-04-10:
Those steps should work, though I've been telling people to do this: https://github.com/marcone/teslausb/wiki/FAQ#q-how-do-i-update-to-the-latest-version-of-teslausb

closed 2020-04-16 by marcone

pgib, 2020-04-16:
Oh, perfect! Not sure how I missed that. Thank you!