marcone/teslausb #465: ~250 snapshots taking 8 minutes to mount

raushel, 2020-04-11:
Last re-built my pi zero (128GB Samsung endurance with 32GB camsize) in early November and have recently updated to 2020.12.5. Currently taking the car around 8 minutes to mount all of the snapshots that currently exist (~250). Today was also the first day I received the "Drive too slow" message since my last re-build of the device.

Going to re-build to resolve (probably without snapshots and AP since the viewer is built into the car now), but wanted to report in case it was useful. I'm not sure the device needs all these snapshots after they've synced as it looks like they're certainly delaying recording availability. Perhaps they could be deleted after they've synced or synced + x number of days has passed.

Looking in diagnostics.txt, today is the first time it's attempted to delete any snapshots due to low space. If a full SD card impacts performance, then perhaps that's causing the too slow messages in the car (too many snapshots --> low space --> low drive performance).

[diagnostics.txt](https://github.com/marcone/teslausb/files/4465532/diagnostics.txt)

marcone, 2020-04-11:
Yes, mounting the snapshots is slow,. I'm surprised you have that many snapshots though. I have about  130, on a 256GB card with a 16G camsize, and they take about 2 minutes to mount.
The purpose of the snapshots is to keep "RecentClips" recordings more than 1 hour and make those recordings available via Samba. They don't (currently) have anything to do with syncing/archiving.

I plan on looking in to using autofs to mount them on-demand, so that they don't delay making the recording drive available to the car.

xbeaudouin, 2020-04-27:
Well, last saturday I got the famous usb too slow.
Is there any possibilty to limit the number of snapshots ?
I really think this slow the IO because of number of snapshots... :(

marcone, 2020-04-27:
> I really think this slow the IO because of number of snapshots... :(

Why do you think that?
People are getting "USB too slow" with regular USB drives, even SSDs, without using teslausb.

raushel, 2020-04-27:
More snapshots --> more storage used --> slower sd card / usb drive  / ssd write. Not sure how the performance impact on an SSD would get down to the level that would be too slow for the car, but usb/sd card can get pretty bad.

Anandtech did some benchmarking around it on ssd's:
https://www.anandtech.com/show/6489/playing-with-op

Anecdotal, but anytime I've gotten usb too slow, wiping and starting fresh has made the too slow errors go away for a few months, at least until snapshots have built up again. Wipe and repeat.

marcone, 2020-04-27:
So how full is your sd card? (What does `df -h /backingfiles` say?) You're using a 128GB card with 32GB camsize. This means that teslausb will try to keep 32GB free, or 25%.
Note that the "spare area" talked about in the article you linked is very different from "free space" on a partition.
All signs point towards this being a Tesla software bug, but since we don't know when Tesla decides to show the "too slow" message, it's all speculation at this point.

marcone, 2020-04-28:
teslausb now uses automount to mount the snapshots, so they are no longer mounted at startup, thus no more delay

closed 2020-04-28 by marcone

xbeaudouin, 2020-04-28:
Thanks for the usage of automount... Well yesterday I have woke up the Tesla to make some update... Just the "df -kh" with about 100 snapshot tooks.... more than 1 minute.
So that's why I really think this part is the one that make TelsaUSB  slowness.
Ok my 512G USB3 key is not the fastest one if I compare to Samsung T5, but it is one that made the system works until the number of snap make this unusable.

marcone, 2020-04-28:
That's why I asked about `df -h /backingfiles` 😄
With just `df -h`, it's going to print the used/free space of every mounted file system, which can obviously take a while if there are hundreds of them.
This has nothing to do with how fast the sd card emulation can access a single file.

Poncherelly, 2020-06-15:
Fyi.

Richard Carpentier
Sr. Product Manager | Managed Digital Solutions
T +1 613 576 2162<tel:+16135762162> | M +1 613 552 8355<tel:+16135528355>

telusinternational.com<https://www.telusinternational.com/>

Connect with us: Twitter<https://twitter.com/TELUSint>  |  YouTube<https://www.youtube.com/user/TELUSInt>  |  LinkedIn<https://www.linkedin.com/company/telus-international/>  |  Facebook<https://www.facebook.com/TELUSInternational>  |  Articles<https://www.telusinternational.com/articles> |  Careers<https://telusinternational.avature.net/careers/Home>

Confidentiality Notice: This e-mail transmission, and any documents, files or previous e-mail messages attached to it may contain confidential information that is legally privileged. If you are not the intended recipient, or a person responsible for delivering it to the intended recipient, you are hereby notified that any disclosure, copying, distribution or use of any of the information contained in or attached to this transmission is STRICTLY PROHIBITED.  If you have received this transmission in error, please immediately notify the sender. Please destroy the original transmission and its attachments without reading or saving in any manner.

From: JEREMY.TAMM@TELUS.COM
Sent: June 15, 2020 6:51 a.m.
To: rob.adams@rubrik.com; timothy.brun@rubrik.com; Farhad.Sadeghi@telus.com
Cc: Paul.Egger@telusinternational.com; Matthew.Wilkes@telus.com; Richard.Carpentier@telus.com
Subject: Data Protection Opportunity Cost

Hi Farhad,

I recently met with Rob and Timothy from Rubrik.  They've expressed a desire to help us more effectively ascertain both the labour improvements and the market opportunities associated with Rubrik in the data protection marketplace.  Richard continues to drive this evaluation but a further deep dive by your team to get a better handle on the productivity opportunity would help.

Rob has shared some insight into how HCL is using Rubrik.  See https://www.rubrik.com/en/resources/webinars/20/hcl-microsoft-hybrid-cloud-strategy.

Please connect with Rob to facilitate a clearer picture of how Rubrik may or may not change your operational experience and effort.  Want to make sure our next discussion on the opportunity for Data Protection Service Evolution here is a robust and fulsome review of our options.

Thank you,

Jeremy
[cid:B1D265E3DB66D6912E253A7CE78CC96F2F0EE7D6@telus.com]
Jeremy Tamm
VP Sales & Marketing| Digital Solutions
T: 1.416.883.8700  M: 1.416.889.4163

telusinternational.com<https://www.telusinternational.com/>
Connect with us: Twitter<https://twitter.com/TELUSint>  |  YouTube<https://www.youtube.com/user/TELUSInt>  |  LinkedIn<https://www.linkedin.com/company/telus-international/>  |  Facebook<https://www.facebook.com/TELUSInternational>  |  Articles<https://www.telusinternational.com/articles> |  Careers<https://telusinternational.avature.net/careers/Home>

Confidentiality Notice: This e-mail transmission, and any documents, files or previous e-mail messages attached to it may contain confidential information that is legally privileged. If you are not the intended recipient, or a person responsible for delivering it to the intended recipient, you are hereby notified that any disclosure, copying, distribution or use of any of the information contained in or attached to this transmission is STRICTLY PROHIBITED.  If you have received this transmission in error, please immediately notify the sender. Please destroy the original transmission and its attachments without reading or saving in any manner.