marcone/teslausb #466: Cannot update - /root/bin/setup-teslausb: line 236: /boot/cmdline.txt: Read-only file system

j450h1, 2020-04-12:
Followed updated instructions to the tee starting with ```sudo -i```. However, I get this error:

```
curl -o /tmp/configure.sh https://raw.githubusercontent.com/marcone/teslausb/585f4105a7df15223a30629e1e9fd1eb38531411/setup/pi/configure.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 11585  100 11585    0     0  47093      0 --:--:-- --:--:-- --:--:-- 47285
Downloaded /tmp/configure.sh ...
~ /boot
Fixing the modules-load parameter in /boot/cmdline.txt...
/root/bin/setup-teslausb: line 236: /boot/cmdline.txt: Read-only file system
```

Any options or do I have to reflash and start from scratch?

marcone, 2020-04-12:
Not really enough information here. Please see https://github.com/marcone/teslausb/issues/35

j450h1, 2020-04-13:
Got it. Here you go. Not sure why there is such a small diagnostics file:

```
root@teslausb:~# cat /tmp/diagnostics.txt 
reading config from /root/teslausb_setup_variables.conf
====== summary ======
hardware: Raspberry Pi Zero W Rev 1.1
OS: Raspbian GNU/Linux 10 (buster)
headless setup config in /root
CIFS archiving selected
```

marcone, 2020-04-13:
Yeah that's weird, there should be a lot more information there. 
Given that, I would suggest starting over.

closed 2020-04-13 by marcone

bit-bash, 2020-06-04:
I know this is closed, but I had this issue as well when I was updating.
Turns out if somehow the "dirty" bit gets set on the /boot partition (like sudden power loss without a clean eject) then when the update scripts tries to mount /boot as rw, the linux file system says no due to the dirty bit and mounts /boot as ro.
I probably caused this myself since I had to do a selfupdate first (older teslausb install), then followed by and upgrade or two.  One of the upgrades hung (no clue why) so I had to yank the power to the Pi which probably left the "dirty" bit set on the /boot partition.