marcone/teslausb #467: With the new in-car Tesla cam viewer, can we configure leaving certain files on the drive?

dogeared, 2020-04-12:
It would be nice to be able to configure a certain date range or a certain amount of file space to leave files on the drive.

Even better: copy everything over, but don't delete from the pi0 - so, a copy instead of a move.

Then, footage could be reviewed both from the network drive and in-car

pretzelpy, 2020-04-16:
I like this idea. Primarily because the in car viewer is a fantastic way to view the different camera angles at the same time.

kevinchau, 2020-04-19:
Suggestion would be to delete the oldest files on the SD card when reaching a certain space threshold.

marcone, 2020-04-19:
Why not just let the car delete the oldest files when it needs space?

kevinchau, 2020-04-19:
> Why not just let the car delete the oldest files when it needs space?

because I forgot that it does that.

aberen001, 2020-04-20:
I vote for adding a setting to allow copy instead of move. That way you'll still have easy access to them in the car, the car can delete older ones, AND you'll have a backup copy on your server just in case the drive gets damaged, stolen, or anything.

sebnaf, 2020-04-28:
> I vote for adding a setting to allow copy instead of move. That way you'll still have easy access to them in the car, the car can delete older vids off the SD card, but you'll still have a backup copy on your server just in case the drive gets damaged, stolen, or anything.

Great idea.

aberen001, 2020-04-28:
The README.md file should probably be updated as well as it currently states:

> You can configure a Raspberry Pi Zero W or Raspberry Pi 4 so that your Tesla thinks it's a USB drive and will write dashcam footage to it. Since it's a computer:
> 
> Scripts running on the Pi can automatically **copy** the clips to an archive server when you get home.

Should probably say **move** instead of copy

simonrb2000, 2020-05-02:
I came on here just to see if this was an option? I have temporarily stopped using TeslaUSB at the moment just because I like the in car viewer and once viewed I would delete them off if nothing interesting. I'd love an option to COPY instead of MOVE. Thanks :) Oh and recently I've been getting the frozen touch screen a lot more often in my X; has anyone else seen this become more regular? Cheers

JR250, 2020-05-11:
Is it safe to just remove the '--remove-source-files' rsync option in 'archive-clips.sh'?

aberen001, 2020-06-09:
Any chance of this @marcone ?

scarymike23, 2020-06-19:
Also interested in this. Another option would be a date range, as in "Copy the files, then delete them x days later (like 30 for example)."

marcone, 2020-06-28:
The "no-delete" branch implements copying the files instead of moving them.
To try it out, add `export BRANCH=no-delete` to your teslausb_setup_variables.conf, then run `setup-teslausb upgrade`

DonC52, 2020-06-29:
Wouldn't the copy (cp -f) continue to copy files that are not deleted in the car to the archive?  If your not keeping the files in the car cleaned up, the copy could take a long time as the number of files keep increasing. You gotta be sure that the archive server can handle the max amount of files that could be copied. Depends on what you have the CAM_SIZE set at.

There are several PC/IOS/Android apps that show the cameras synced together (Like https://sentrycam.appspot.com/ ) among others.

marcone, 2020-06-29:
> Wouldn't the copy (cp -f) continue to copy files that are not deleted in the car to the archive?

No, because before that there's a check to see if the destination file already exists.

aberen001, 2020-06-29:
Brilliant! Thank you so much for doing this, downloading it tonight!!

DonC52, 2020-06-29:
> 
> 
> > Wouldn't the copy (cp -f) continue to copy files that are not deleted in the car to the archive?
> 
> No, because before that there's a check to see if the destination file already exists.

Oops - didn't look far enough up.  I take it that Linux doesn't keep partial copied files if power is lost or is somehow interrupted? (Maybe an Archive OS question really.)  Guess that the user could just delete a partial copied file and get it the next time.

scarymike23, 2020-07-07:
Anyone using this? I've been using it for over a week and I like the functionality. HOWEVER, I get many alerts via the IFTTT integration that its moved files, when in actuality it just checked, saw the files already existed, and didn't move anything. I'm guessing a few tweaks are needed to prevent that.

scarymike23, 2020-07-09:
Here is the notice I get, pretty much every time the car is taken out of sleep mode (and I'm at home). 

[image]

DonC52, 2020-07-09:
That message is sent in the archiveloop before it looks at rather they are already archived or not. Take a look at the log file in /mutable and you will most likely see that the files already archived are being skipped.

jackbrown1993, 2020-12-12:
@marcone is there a reason this isn't merged into the main branch? I can see the no-delete branch hasn't been rebased for a while.

marcone, 2020-12-12:
@jackbrown1993 it turns out that the car will never delete manually saved clips, so without delete-after-archive the drive can fill up and it will stop recording. Also, notifications aren't working right as people have noted above.

miles267, 2020-12-12:
> @jackbrown1993 it turns out that the car will never delete manually saved clips, so without delete-after-archive the drive can fill up and it will stop recording. Also, notifications aren't working right as people have noted above.

How do you enable this delete-after-archive option? I’ve been receiving usb getting full messages on 2020.48.10. TeslaUSB has been consistently archiving to my home file server. Turns out, when I open the video viewer in the car, I’m seeing many videos over a few months that haven’t been deleted for whatever reason. I can certain go in and manually delete these periodically if needed, but wasn’t sure why they’re still there?

marcone, 2020-12-12:
> How do you enable this delete-after-archive option?

It is enabled by default (unless you're using the "no-delete" branch).

shawnbissell, 2020-12-22:
Since #580 was closed in favor of this issue, I will repeat my suggestion here. 

For the "no-delete" branch/option to be truly accepted, there needs to be an option to delete archived footage after X days. A default of 30 days would probably work for 95% of the people with a 64GB card or higher. Ideally it would work like the delete-after-archive process works only deletes files that it knows have been successfully archived.

shawnbissell, 2020-12-23:
I've created a branch here https://github.com/shawnbissell/teslausb that has a new single option called NUM_DAYS_TO_KEEP which when set to a number > 0 will delete files older than that many days after archiving them. This branch is forked from main-dev and is not dependent on the changes in the "no-cache" branch. Currently NUM_DAYS_TO_KEEP does not work for rsync or rclone. I'll  create a proper PR once I've figured out how to support them.

psqv, 2020-12-24:
> I've created a branch here https://github.com/shawnbissell/teslausb that has a new single option called NUM_DAYS_TO_KEEP which when set to a number > 0 will delete files older than that many days [...]

Splendid! My drive has been filling up with the no-delete branch. Can I somehow use this branch by editing the "export BRANCH=" property in teslausb_setup_variables.conf?

shawnbissell, 2020-12-26:
@psqv yes you need to change the repo like this
`export REPO=shawnbissell`

shawnbissell, 2020-12-29:
@marcone not sure if you have had a look at my branch yet, but I'm guessing all the timestamp and deletion should be moved into the archiveloop.sh to avoid repeating it in each of the archive-clips.sh files. Right after the "/root/bin/archive-clips.sh" call. Does that make the most sense to you?

shawnbissell, 2020-12-29:
On second thought if the code is centralized in the archiveloop there is no way to know if a file was successfully copied or not ... just blindingly deleting the old files is a bad idea ... not sure what to do now

marcone, 2021-01-05:
@shawnbissell I was thinking of doing something like:
- create an [overlay mount](https://www.kernel.org/doc/html/latest/filesystems/overlayfs.html) with the cam drive as the "lower" and a tmpfs dir as the "upper" (*)
- have archive-clips.sh archive the files as usual, but from the overlay
- since it's using an overlay, the deletes are only marked in the "upper" dir, but don't propagate to the actual cam drive
- archiveloop then determines which files were archived by examining the overlay after archiving completes, and does the actual file deletion in the underlying cam drive based on archive status, age, available space, etc.
- archiveloop should keep a list of previously-archived files, and exclude those files from the next archive cycle, so there's no additional work/logic necessary in any of the archive-clips scripts
I did some experiments with the above and it seems like it will work.

(*) overlayfs doesn't directly work with a vfat fs as the lower, but that can be worked around by first mounting a [posixovl ](http://manpages.ubuntu.com/manpages/bionic/man1/mount.posixovl.1.html)on top of the cam drive, and then using the posixovl as the lower for the overlayfs mount

MarcFinns, 2021-01-08:
> Wouldn't the copy (cp -f) continue to copy files that are not deleted in the car to the archive? If your not keeping the files in the car cleaned up, the copy could take a long time as the number of [...]

Is there a list of those viewers? thanks!

shawnbissell, 2021-01-18:
@marcone sorry for not responding earlier ... sounds like a good plan! Let me know if you want me to test it out.

marcone, 2021-03-28:
This is now implemented

closed 2021-03-28 by marcone