marcone/teslausb #468: Make use of new Json event metadata

washcroft, 2020-04-15:
Sentry mode in 2020.12.5 now has an event file with the reason, timestamp and location recorded:

[image]

Would this file also be archived alongside footage?

marcone, 2020-04-15:
It should be archived for rclone and rsync, but cifs only archives files greater than 100K, so won't archive these. That needs to be fixed.

marcone, 2020-04-18:
the json file is now archived for cifs too

closed 2020-04-18 by marcone

GoTeslaGo, 2020-04-18:
Awesome. Anything we have to do to get this to work? I’m running 20190615 so would a selfupdate resolve the issue for me or do I need to use a later version of TeslaUSB?

marcone, 2020-04-18:
https://github.com/marcone/teslausb/wiki/FAQ#q-how-do-i-update-to-the-latest-version-of-teslausb