marcone/teslausb #469: How can I get rPi to record to SSD?

colinstalter, 2020-04-17:
I don't have internet access in my garage so I would like the rPi to archive the recordings onto an SSD (which is plugged into the rPi) instead of to the microSD card.

What is the best way to set this up?

marcone, 2020-04-17:
There is no way to archive to an SSD, but you can just plug the SSD in to the car directly (using a SATA to USB adapter), no Pi required.

colinstalter, 2020-04-17:
I guess what I'm saying is I want the footage to be recorded to an SSD
plugged into the Pi instead of to the PI's microSD. The problem with just
plugging the SSD into the tesla is it only saves the last 60 minutes of
footage before looping around.

I want to use your program to archive dump the footage to a separate folder
before tesla deletes it.

marcone, 2020-04-17:
teslausb already keeps more than the last hour of footage. You can either use a regular sd card for that (i.e. the standard setup), or a Pi4 with a hard drive or SSD. It's as easy as specifying [the right option](https://github.com/marcone/teslausb/blob/main-dev/pi-gen-sources/00-teslausb-tweaks/files/teslausb_setup_variables.conf.sample#L23) in the configuration file.

However, that relies on accessing the Pi via wifi to then access those recordings, so since you say you have no wifi where you park, you'll have to bring the Pi in to the house (or wherever you have wifi) to access the files.

closed 2020-04-17 by marcone

colinstalter, 2020-04-17:
Thanks, thanks fine with me. I just want to be able to have it record footage until the 1TB is full. Right now it only records the last 60 minutes and then loops back around.