marcone/teslausb #470: Sync Music folder

Ramdg33, 2020-04-19:
For unclear reason my local music folder isn't sync'd with my music folder on tesla usb while the rest is all goog. I have reserved 100G and when I connect it to my iMac both the CAM and MUSIC folder show up and I'm also able to add music to the folder. When connecting the Pi to my tesla I can select the music from the music folder. I have done new installs and tried different cables. Is there a way to remotely connect to the music folder in the same way as TeslaCam?

marcone, 2020-04-19:
> For unclear reason  ...

Looking at the diagnostics would probably make this more clear. See https://github.com/marcone/teslausb/issues/35

closed 2020-04-19 by marcone

Ramdg33, 2020-04-19:
At this point I'm more interested to know if I can remotely access the music folder. Could you provide some direction?

marcone, 2020-04-19:
That is not currently supported.

Ramdg33, 2020-04-20:
ok