marcone/teslausb #472: Fails to move/copy files to SMB shares

cleanev, 2020-04-22:
I made new TeslaUSB using prebuilt image and letting it go thru motions. Somehow even though there are many events recorded in Recent/Saved and Sentry folders on USB (pi) they do not get moved over.
Tesla Model 3 (2018 RWD LR) with 2020.12.5 e2179e0650f0 update

Below is the latest log
Wed 22 Apr 12:42:57 EDT 2020: Starting archiveloop...
Wed 22 Apr 12:42:57 EDT 2020: mounting snapshot /backingfiles/snapshots/snap-000000/snap.bin
Wed 22 Apr 12:43:00 EDT 2020: mounting snapshot /backingfiles/snapshots/snap-000001/snap.bin
Wed 22 Apr 12:43:02 EDT 2020: mounting snapshot /backingfiles/snapshots/snap-000002/snap.bin
Wed 22 Apr 12:43:03 EDT 2020: mounted all snapshots
Wed 22 Apr 12:43:04 EDT 2020: taking snapshot of cam disk in /backingfiles/snapshots/snap-000003
Wed 22 Apr 12:43:09 EDT 2020: took snapshot
Wed 22 Apr 12:43:11 EDT 2020: comparing new snapshot with /backingfiles/snapshots/snap-000002/snap.bin
Wed 22 Apr 12:43:11 EDT 2020: new snapshot is identical to previous one, discarding
Wed 22 Apr 12:43:12 EDT 2020: Running fsck on /backingfiles/cam_disk.bin...
Wed 22 Apr 12:43:12 EDT 2020: | fsck from util-linux 2.33.1
Wed 22 Apr 12:43:15 EDT 2020: | fsck.fat 4.1 (2017-01-24)
Wed 22 Apr 12:43:15 EDT 2020: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Wed 22 Apr 12:43:15 EDT 2020: |  Automatically removing dirty bit.
Wed 22 Apr 12:43:15 EDT 2020: | Performing changes.
Wed 22 Apr 12:43:15 EDT 2020: | /dev/loop3p1: 287 files, 550024/2094976 clusters
Wed 22 Apr 12:43:15 EDT 2020: Finished fsck on /backingfiles/cam_disk.bin.
Wed 22 Apr 12:43:15 EDT 2020: Connecting usb to host...
Wed 22 Apr 12:43:15 EDT 2020: Connected usb to host.
Wed 22 Apr 12:43:15 EDT 2020: Waiting for archive to be reachable...
Wed 22 Apr 13:38:38 EDT 2020: Archive is reachable.
Wed 22 Apr 13:38:38 EDT 2020: Waiting for time to be set by ntpd...
Wed 22 Apr 13:39:05 EDT 2020: Time still not set, attempting to force it
Wed 22 Apr 13:59:03 EDT 2020: taking snapshot of cam disk in /backingfiles/snapshots/snap-000003
Wed 22 Apr 13:59:06 EDT 2020: took snapshot
Wed 22 Apr 13:59:09 EDT 2020: comparing new snapshot with /backingfiles/snapshots/snap-000002/snap.bin
Wed 22 Apr 13:59:09 EDT 2020: making links for /tmp/snapshots/snap-000003, retargeted to /backingfiles/snapshots/snap-000003/mnt
Wed 22 Apr 13:59:14 EDT 2020: made all links for /tmp/snapshots/snap-000003
Wed 22 Apr 13:59:14 EDT 2020: Disconnecting usb from host...
Wed 22 Apr 13:59:15 EDT 2020: Disconnected usb from host.
Wed 22 Apr 13:59:15 EDT 2020: Running fsck on /backingfiles/cam_disk.bin...
Wed 22 Apr 13:59:15 EDT 2020: | fsck from util-linux 2.33.1
Wed 22 Apr 13:59:17 EDT 2020: | fsck.fat 4.1 (2017-01-24)
Wed 22 Apr 13:59:17 EDT 2020: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Wed 22 Apr 13:59:17 EDT 2020: |  Automatically removing dirty bit.
Wed 22 Apr 13:59:17 EDT 2020: | Performing changes.
Wed 22 Apr 13:59:17 EDT 2020: | /dev/loop4p1: 321 files, 1051774/2094976 clusters
Wed 22 Apr 13:59:17 EDT 2020: Finished fsck on /backingfiles/cam_disk.bin.
Wed 22 Apr 13:59:17 EDT 2020: Archiving...
Wed 22 Apr 13:59:17 EDT 2020: Ensuring cam archive is mounted...
Wed 22 Apr 13:59:17 EDT 2020: Mounting /mnt/archive...
Wed 22 Apr 13:59:18 EDT 2020: Mounted /mnt/archive.
Wed 22 Apr 13:59:18 EDT 2020: Ensured cam archive is mounted.
Wed 22 Apr 13:59:18 EDT 2020: Checking saved folder count...
Wed 22 Apr 13:59:18 EDT 2020: Ensuring cam file is mounted...
Wed 22 Apr 13:59:18 EDT 2020: Mounting /mnt/cam...
Wed 22 Apr 13:59:18 EDT 2020: Mounted /mnt/cam.
Wed 22 Apr 13:59:18 EDT 2020: Ensured cam file is mounted.
Wed 22 Apr 13:59:20 EDT 2020: There are 3 event folder(s) with 311 file(s) to move.
Wed 22 Apr 13:59:20 EDT 2020: Starting recording archiving: 3 event folder(s) with 311 file(s)
Wed 22 Apr 13:59:21 EDT 2020: Moving clips to archive...
Wed 22 Apr 13:59:21 EDT 2020: Creating output directory 'SavedClips/2020-04-22_10-57-40'
Wed 22 Apr 13:59:21 EDT 2020: Moving 'SavedClips/2020-04-22_10-57-40/2020-04-22_10-46-59-front.mp4'
Wed 22 Apr 14:00:21 EDT 2020: Failed to move 'SavedClips/2020-04-22_10-57-40/2020-04-22_10-46-59-front.mp4'
Wed 22 Apr 14:00:21 EDT 2020: Creating output directory 'SavedClips/2020-04-22_10-57-40'
Wed 22 Apr 14:00:21 EDT 2020: Moving 'SavedClips/2020-04-22_10-57-40/2020-04-22_10-46-59-left_repeater.mp4'
Wed 22 Apr 14:00:41 EDT 2020: waiting up to 90 seconds for idle interval
Wed 22 Apr 14:00:42 EDT 2020: mass storage process not active, OK to write
Wed 22 Apr 14:00:42 EDT 2020: taking snapshot of cam disk in /backingfiles/snapshots/snap-000004
Wed 22 Apr 14:00:46 EDT 2020: took snapshot
Wed 22 Apr 14:00:48 EDT 2020: comparing new snapshot with /backingfiles/snapshots/snap-000003/snap.bin
Wed 22 Apr 14:00:48 EDT 2020: new snapshot is identical to previous one, discarding
Wed 22 Apr 14:38:05 EDT 2020: Failed to move 'SavedClips/2020-04-22_10-57-40/2020-04-22_10-46-59-left_repeater.mp4'
Wed 22 Apr 14:38:05 EDT 2020: Creating output directory 'SavedClips/2020-04-22_10-57-40'
Wed 22 Apr 14:38:05 EDT 2020: Moving 'SavedClips/2020-04-22_10-57-40/2020-04-22_10-46-59-right_repeater.mp4'
Wed 22 Apr 14:58:49 EDT 2020: waiting up to 90 seconds for idle interval
Wed 22 Apr 14:58:50 EDT 2020: mass storage process not active, OK to write
Wed 22 Apr 14:58:50 EDT 2020: taking snapshot of cam disk in /backingfiles/snapshots/snap-000004
Wed 22 Apr 14:58:54 EDT 2020: took snapshot
Wed 22 Apr 14:58:56 EDT 2020: comparing new snapshot with /backingfiles/snapshots/snap-000003/snap.bin
Wed 22 Apr 14:58:56 EDT 2020: new snapshot is identical to previous one, discarding

marcone, 2020-04-22:
This part is really strange:
```
Wed 22 Apr 14:00:21 EDT 2020: Moving 'SavedClips/2020-04-22_10-57-40/2020-04-22_10-46-59-left_repeater.mp4'
Wed 22 Apr 14:00:41 EDT 2020: waiting up to 90 seconds for idle interval
Wed 22 Apr 14:00:42 EDT 2020: mass storage process not active, OK to write
Wed 22 Apr 14:00:42 EDT 2020: taking snapshot of cam disk in /backingfiles/snapshots/snap-000004
Wed 22 Apr 14:00:46 EDT 2020: took snapshot
Wed 22 Apr 14:00:48 EDT 2020: comparing new snapshot with /backingfiles/snapshots/snap-000003/snap.bin
Wed 22 Apr 14:00:48 EDT 2020: new snapshot is identical to previous one, discarding
Wed 22 Apr 14:38:05 EDT 2020: Failed to move 'SavedClips/2020-04-22_10-57-40/2020-04-22_10-46-59-left_repeater.mp4'
```
as it seems to indicate that it tried for 38 minutes to move 1 file, then failed.
I suppose it's possible for it to take that long if your server or wifi is extremely slow...

cleanev, 2020-04-22:
Hi @marcone -
I would come to same conclusion. However the disk in question is an SSD cluster hosted on HPDL380 and has worked flawlessly till date. I saw this behavior first time in many iterations you have published so wondered if any change to code may have contributed to this uptick?

I will recheck all the settings on server. Garage and server/router location has solid signal as they are just a wall apart. I also have full AIMesh (ASUS) with 4 routers (super overkill) to make sure there is not one area left without signal.

marcone, 2020-04-22:
That's not a log, that's a directory listing.
It would probably be more helpful if you followed the instructions for filing issues in https://github.com/marcone/teslausb/issues/35

cleanev, 2020-04-22:
Sorry about that @marcone. I ran all 3 commands and created a new paste in account where I have posted the diagnostics log 
https://pastebin.com/Evhr6J76

marcone, 2020-04-22:
From the diagnostics, it looks like it's working now:
```
Wed 22 Apr 17:55:53 EDT 2020: Moving 'SavedClips/2020-04-22_10-57-40/2020-04-22_10-51-00-back.mp4'
Wed 22 Apr 17:56:13 EDT 2020: Moved 'SavedClips/2020-04-22_10-57-40/2020-04-22_10-51-00-back.mp4'
Wed 22 Apr 17:56:13 EDT 2020: Creating output directory 'SavedClips/2020-04-22_10-57-40'
Wed 22 Apr 17:56:13 EDT 2020: Moving 'SavedClips/2020-04-22_10-57-40/2020-04-22_10-52-01-front.mp4'
Wed 22 Apr 17:56:33 EDT 2020: Moved 'SavedClips/2020-04-22_10-57-40/2020-04-22_10-52-01-front.mp4'
Wed 22 Apr 17:56:33 EDT 2020: Creating output directory 'SavedClips/2020-04-22_10-57-40'
Wed 22 Apr 17:56:33 EDT 2020: Moving 'SavedClips/2020-04-22_10-57-40/2020-04-22_10-52-01-left_repeater.mp4'
Wed 22 Apr 17:56:54 EDT 2020: Moved 'SavedClips/2020-04-22_10-57-40/2020-04-22_10-52-01-left_repeater.mp4'
```

It's a little slow, about 20 seconds per file, so expect about 1.5-2 hours before archiving is done.

cleanev, 2020-04-22:
Thanks @marcone. I see it finished. In the meantime I have made new USB drive that I will plugin tomorrow to see how that fares

closed 2020-04-22 by cleanev