marcone/teslausb #473: TeslaUsb not always recognized by the car

taugusti, 2020-04-25:
Hi,
Not sure where the problem is... but I have used TeslaUsb for 8 months now - and from time to time (maybe every 3rd to 4th trip) the pi zero and TeslaUsb are not recognized by the car.

I recently updated to version 20200122 in the hope that the problem would go away. Unfortunately no luck.

The pi zero is connected through a 5 cm long usb-cable but I have tried with different cables without any changes experienced.

I have two friends having the same problem - so it does not seen to be related to a fault in my pi zero. But... since I am not getting any hits on google for the problem, I suspect it may be something we all missed during installation?

I hope someone can help me identify the problem...

Thanks in advance

marcone, 2020-04-25:
Not enough information here. See https://github.com/marcone/teslausb/issues/35

taugusti, 2020-04-25:
> Not enough information here. See #35

Added diagnostics.txt

marcone, 2020-04-25:
Were these diagnostics taken when the Pi was in the unrecognized-by-the-car state, or was the car off, or the Pi not connected to anything? (asking because it looks like the Pi had made the drive available, but nothing had accessed it)
When the car doesn't see the Pi, does the Pi still blink its LED? Can you ssh in to it?

taugusti, 2020-04-25:
> Were these diagnostics taken when the Pi was in the unrecognized-by-the-car state, or was the car off, or the Pi not connected to anything? (asking because it looks like the Pi had made the drive available, but nothing had accessed it)
I do not know for sure if it was working or not when I got the file. I just woke the car via the tesla app and ssh'ed into it when it came online. I expected that the diagnostics file went days back - but apparently that is not the case.

> When the car doesn't see the Pi, does the Pi still blink its LED? Can you ssh in to it?
I will check this next time I see the problem. I will get back as soon as I have investigated it.

taugusti, 2020-04-25:
New diagnostics.txt file: https://pastebin.com/fWs23WGw
pi zero is blinking two flashes - pause - repeat
teslausb is available on both samba and ssh

I hope this brings some light over the matter...?

marcone, 2020-04-25:
Just to be clear, you took these diagnostics while the car's screen was on, but it wasn't showing the dashcam icon?

The two blinks are the normal mode of operation when the Pi can connect to the archive server.
The diagnostics show the Pi publishing the drive, but nothing is connected it. Usually this means a bad (or charge-only) USB cable.

taugusti, 2020-04-25:
> Just to be clear, you took these diagnostics while the car's screen was on, but it wasn't showing the dashcam icon?

yes. screen on, no dashcan icon.

> The two blinks are the normal mode of operation when the Pi can connect to the archive server.
> The diagnostics show the Pi publishing the drive, but nothing is connected it. Usually this means a bad (or charge-only) USB cable.

Hmm. It works most of the time. But - I have previously had issues with the cables when I started using teslausb. So it might be something with the cable.
But why are they so special for this? The cables i have tried have been used for powering other stuff and for connecting various usb devices to a pc - like a usb ssd drive. But they still do not work for the pi zero...?

How do I pick the right cable?

marcone, 2020-04-25:
There's nothing special about the required cable. All I'm saying is that the symptoms you're experiencing are consistent with using a charge-only cable. It could be you have a break in the cable somewhere that makes it so it occasionally loses contact and that's why it only happens some times. It could be the USB port on the Pi or in the car is loose.
There are probably other explanations as well, but I see nothing wrong in your diagnostics; as far as the Pi is concerned, the drive is ready to be accessed, but nothing is accessing it.

taugusti, 2020-04-26:
Ok. I will check for bad solderings and swap to the other port in the car.

marcone, 2020-05-01:
any luck?

taugusti, 2020-05-02:
Hi,
Too bad the pi-zero is so sensitive to cable.
I decided to buy the small usb adaptor that can be soldered directly to the pi-zero.
and so far it seems to work - but... I have not been driving much since it arrived (corona stay-at-home policy).
But - my feeling is that it has helped. 
I will leave better feedback net week when i have tested some more.

But  in any case thanks for your quick and thorough help!
/th

taugusti, 2020-05-06:
Hi,
As mentioned I soldered the usb-stem on the pi zero - and I have had 100% success since!
So - I will close the thread.
And again: Thanks for your support!

closed 2020-05-06 by taugusti