marcone/teslausb #474: Not working with v10.2

ubehera, 2020-04-26:
Was working perfectly earlier, but I recently got FSD upgrade. It’s not being recognized by the car neither for dashcam, nor music since then. I tried reflashing latest code, but still the same issue. Switched to the other USB port, still no luck

marcone, 2020-04-26:
Not enough information. See https://github.com/marcone/teslausb/issues/35

jun3280net, 2020-04-26:
> Was working perfectly earlier, but I recently got FSD upgrade. It’s not being recognized by the car neither for dashcam, nor music since then. I tried reflashing latest code, but still the same issue. Switched to the other USB port, still no luck

Your issue might simply be related to the version of software you are on : 
[https://github.com/marcone/teslausb/issues/450](https://github.com/marcone/teslausb/issues/450)

Have you tried using a basic USB formatted with FAT32 for music or dashcam?

ubehera, 2020-04-26:
> Not enough information. See #35
PFA the diagnostics file. Also added screenshot of the version I'm running, I don't think it's related to the S/X problem.
[image]
 per #35 
[diagnostics.txt](https://github.com/marcone/teslausb/files/4536088/diagnostics.txt)

ubehera, 2020-04-26:
> Not enough information. See #35

PFA diagnostics file and my current version number. I don't think this is related to the S/X issue.
[diagnostics.txt](https://github.com/marcone/teslausb/files/4536114/diagnostics.txt)
[image]

ubehera, 2020-04-26:
> Have you tried using a basic USB formatted with FAT32 for music or dashcam?

Not yet, let me try to find a stick, I'll try it ASAP.

jun3280net, 2020-04-26:
All good man.  I think with all these software updates Tesla has been
pushing through, its been breaking things.  They have too many branches of
configurations right now.  The basic USB setup is the quickest and easiest
thing to test before spending too much time debugging TeslaUsb.  Good luck!

marcone, 2020-04-26:
Can you wait a little longer after boot before taking the diagnostics? In the diagnostics you attached, it was still trying to copy music from your server and hadn't made the drives accessible to the car/PC yet.
(and there seemed to be some problem contacting your server: `CIFS VFS: SMB signature verification returned error = -5`)

ubehera, 2020-04-26:
Checked with USB, it works. I’ll pull logs again tonight.

ubehera, 2020-04-27:
> Can you wait a little longer after boot before taking the diagnostics? In the diagnostics you attached, it was still trying to copy music from your server and hadn't made the drives accessible to [...]

@marcone attaching 2 log files, (1) was collected after a drive, and (2) was collected after a clean reboot in the garage.
[diagnostics-2.txt](https://github.com/marcone/teslausb/files/4536888/diagnostics-2.txt)
[diagnostics-1.txt](https://github.com/marcone/teslausb/files/4536889/diagnostics-1.txt)

marcone, 2020-04-27:
Your diagnostics are consistent with using a bad (or charge-only) USB cable.
You also seem to be having some issues with your archive server, as noted above.
Other than that, nothing stands out to me. Have you tried connecting the Pi to a PC?

ubehera, 2020-04-27:
> Your diagnostics are consistent with using a bad (or charge-only) USB cable.
> 
> You also seem to be having some issues with your archive server, as noted above.
> 
> Other than that, nothing stands out to me. Have you tried connecting the Pi to a PC?
> 
> 
> 
> 

Yep. Works fine on PC. Also, nothing changed on the setup HW wise, it’s the same cable and same board. Only new FW on the car and new TeslaUSB version.

marcone, 2020-04-28:
If it works with a PC, and you're using the same cable with the car, then the problem is with the car. From the Pi/teslausb's point of view, there is no difference between being connected to a car or to a PC.

closed 2020-04-28 by marcone

ubehera, 2020-04-28:
@marcone Question, is there anything specific you want me to do on the PC to verify it’s working as intended. When I said it works “fine” I meant it’s booting up normally, not sure how to test the USB portion. Also, a normal USB stick is working in the car. I’ll try it again on PC and see if it shows up as a partition.

marcone, 2020-04-28:
ah, when you said that it works with a PC I thought you meant that the cam/music drives show up on the PC.

ubehera, 2020-05-01:
I can confirm that the USB drives don’t show up on PC or Mac.

ubehera, 2020-05-01:
I’ll try an older release and check.

marcone, 2020-05-01:
So far all signs point to a bad cable, or using the wrong USB port on the Pi.

ubehera, 2020-05-01:
I’m go to a corner and cry now. Sorry for wasting everyone’s time. I had it plugged into the power only port. I forgot that the case I use has them marked wrong. Apologies.