marcone/teslausb #475: STOP: The MicroSD card is too small: 12601262080 bytes available.

m3talstorm, 2020-04-27:
- teslausb-20200122 (Etcher)
- Empty (all partitions deleted) 16GB FD
- camsize has been commented out

```pi@teslausb:~ $ cat /boot/teslausb-headless-setup.log
Mon 27 Apr 15:38:34 BST 2020 : Detecting whether to update wpa_supplicant.conf
Mon 27 Apr 15:38:34 BST 2020 : Starting setup.
Mon 27 Apr 15:38:35 BST 2020 : reading config from /root/teslausb_setup_variables.conf
Mon 27 Apr 15:38:35 BST 2020 : curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev
Mon 27 Apr 15:38:35 BST 2020 : curl -o /tmp/setup-teslausb https://raw.githubusercontent.com/marcone/teslausb/f999aaeb52e7201e94e02dcc12299aa72d5e2f2a/setup/pi/setup-teslausb
Mon 27 Apr 15:38:36 BST 2020 : Downloaded /tmp/setup-teslausb ...
Mon 27 Apr 15:38:36 BST 2020 : /root/bin/setup-teslausb is up to date
Mon 27 Apr 15:38:36 BST 2020 : Updating package index files...
Mon 27 Apr 16:13:55 BST 2020 : Verifying that the requested configuration is valid...
Mon 27 Apr 16:13:55 BST 2020 : curl -o /tmp/verify-configuration.sh https://raw.githubusercontent.com/marcone/teslausb/f999aaeb52e7201e94e02dcc12299aa72d5e2f2a/setup/pi/verify-configuration.sh
Mon 27 Apr 16:13:56 BST 2020 : Downloaded /tmp/verify-configuration.sh ...
Mon 27 Apr 16:13:56 BST 2020 : usb_drive is not set. SD card will be used.
Mon 27 Apr 16:13:56 BST 2020 : Verifying that there is sufficient space available on the MicroSD card...
Mon 27 Apr 16:13:58 BST 2020 : STOP: The MicroSD card is too small: 12601262080 bytes available.
Mon 27 Apr 16:13:59 BST 2020 : Model: SD SL16G (sd/mmc)
Disk /dev/mmcblk0: 15.9GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start   End     Size    Type     File system  Flags
 1      4194kB  273MB   268MB   primary  fat32        lba
 2      273MB   3334MB  3062MB  primary  ext4
```

What am I doing wrong?

marcone, 2020-04-27:
Did you specify an expanded root partition in the config file? The root partition isn't normally 3GB, and because of this, you now only have 11.7GB remaining, whereas teslausb wants at least 12 GB.

m3talstorm, 2020-04-27:
I changed it to 1G at the very very start, never 3G, I had rebooted a few times because I saw the same message every time, is it growing it each time? 

I'll wipe it all again and try again.

marcone, 2020-04-27:
The size you specify in the config file is the *additional* space setup adds to the root partition. It only does this once.
The root partition is about 2GB by default, so with the additional 1GB on top of that it didn't leave enough space for teslausb.

closed 2020-04-27 by marcone