marcone/teslausb #476: Archiving does not complete

cleanev, 2020-04-28:
Why is archiving stopping in latest build?

Archivelog snippet since parking at home. I am connected to Pi over SSH as the car and therefore the Pi are both active. Still only few files got copied over from 248.

Mon 27 Apr 19:54:46 EDT 2020: Time still not set, attempting to force it
Mon 27 Apr 19:55:19 EDT 2020: taking snapshot of cam disk in /backingfiles/snapshots/snap-000060
Mon 27 Apr 19:55:22 EDT 2020: took snapshot
Mon 27 Apr 19:55:25 EDT 2020: comparing new snapshot with /backingfiles/snapshots/snap-000059/snap.bin
Mon 27 Apr 19:55:25 EDT 2020: making links for /tmp/snapshots/snap-000060, retargeted to /backingfiles/snapshots/snap-000060/mnt
Mon 27 Apr 19:55:29 EDT 2020: made all links for /tmp/snapshots/snap-000060
Mon 27 Apr 19:55:29 EDT 2020: Disconnecting usb from host...
Mon 27 Apr 19:55:29 EDT 2020: Disconnected usb from host.
Mon 27 Apr 19:55:29 EDT 2020: Running fsck on /backingfiles/cam_disk.bin...
Mon 27 Apr 19:55:30 EDT 2020: | fsck from util-linux 2.33.1
Mon 27 Apr 19:55:31 EDT 2020: | fsck.fat 4.1 (2017-01-24)
Mon 27 Apr 19:55:31 EDT 2020: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Mon 27 Apr 19:55:32 EDT 2020: |  Automatically removing dirty bit.
Mon 27 Apr 19:55:32 EDT 2020: | Long filename fragment "2020-04-27_17" found outside a LFN sequence.
Mon 27 Apr 19:55:32 EDT 2020: |   (Maybe the start bit is missing on the last fragment)
Mon 27 Apr 19:55:32 EDT 2020: |   Not auto-correcting this.
Mon 27 Apr 19:55:32 EDT 2020: | Long filename fragment "2020-04-26_17" found outside a LFN sequence.
Mon 27 Apr 19:55:32 EDT 2020: |   (Maybe the start bit is missing on the last fragment)
Mon 27 Apr 19:55:32 EDT 2020: |   Not auto-correcting this.
Mon 27 Apr 19:55:32 EDT 2020: | Long filename fragment "2020-04-25_10" found outside a LFN sequence.
Mon 27 Apr 19:55:32 EDT 2020: |   (Maybe the start bit is missing on the last fragment)
Mon 27 Apr 19:55:32 EDT 2020: |   Not auto-correcting this.
Mon 27 Apr 19:55:32 EDT 2020: | Long filename fragment "2020-04-23_17" found outside a LFN sequence.
Mon 27 Apr 19:55:32 EDT 2020: |   (Maybe the start bit is missing on the last fragment)
Mon 27 Apr 19:55:32 EDT 2020: |   Not auto-correcting this.
Mon 27 Apr 19:55:32 EDT 2020: | Long filename fragment "2020-04-23_17" found outside a LFN sequence.
Mon 27 Apr 19:55:32 EDT 2020: |   (Maybe the start bit is missing on the last fragment)
Mon 27 Apr 19:55:32 EDT 2020: |   Not auto-correcting this.
Mon 27 Apr 19:55:32 EDT 2020: | Performing changes.
Mon 27 Apr 19:55:32 EDT 2020: | /dev/loop9p1: 269 files, 785381/2094976 clusters
Mon 27 Apr 19:55:32 EDT 2020: Finished fsck on /backingfiles/cam_disk.bin.
Mon 27 Apr 19:55:32 EDT 2020: Archiving...
Mon 27 Apr 19:55:32 EDT 2020: Ensuring cam archive is mounted...
Mon 27 Apr 19:55:32 EDT 2020: Mounting /mnt/archive...
Mon 27 Apr 19:55:33 EDT 2020: Mounted /mnt/archive.
Mon 27 Apr 19:55:33 EDT 2020: Ensured cam archive is mounted.
Mon 27 Apr 19:55:33 EDT 2020: Checking saved folder count...
Mon 27 Apr 19:55:33 EDT 2020: Ensuring cam file is mounted...
Mon 27 Apr 19:55:33 EDT 2020: Mounting /mnt/cam...
Mon 27 Apr 19:55:33 EDT 2020: Mounted /mnt/cam.
Mon 27 Apr 19:55:33 EDT 2020: Ensured cam file is mounted.
Mon 27 Apr 19:55:35 EDT 2020: There are 4 event folder(s) with 258 file(s) to move.
Mon 27 Apr 19:55:35 EDT 2020: Starting recording archiving: 4 event folder(s) with 258 file(s)
Mon 27 Apr 19:55:35 EDT 2020: Moving clips to archive...
Mon 27 Apr 19:55:36 EDT 2020: Creating output directory 'SentryClips/2020-04-27_14-17-25'
Mon 27 Apr 19:55:52 EDT 2020: Moving 'SentryClips/2020-04-27_14-17-25/2020-04-27_14-17-23-front.mp4'
Mon 27 Apr 19:56:33 EDT 2020: Moved 'SentryClips/2020-04-27_14-17-25/2020-04-27_14-17-23-front.mp4'
Mon 27 Apr 19:56:33 EDT 2020: Creating output directory 'SentryClips/2020-04-27_14-17-25'
Mon 27 Apr 19:56:33 EDT 2020: Moving 'SentryClips/2020-04-27_14-17-25/2020-04-27_14-17-23-left_repeater.mp4'
Mon 27 Apr 19:56:59 EDT 2020: Moved 'SentryClips/2020-04-27_14-17-25/2020-04-27_14-17-23-left_repeater.mp4'
Mon 27 Apr 19:56:59 EDT 2020: Creating output directory 'SentryClips/2020-04-27_14-17-25'
Mon 27 Apr 19:56:59 EDT 2020: Moving 'SentryClips/2020-04-27_14-17-25/2020-04-27_14-17-23-right_repeater.mp4'
Mon 27 Apr 19:57:02 EDT 2020: Moved 'SentryClips/2020-04-27_14-17-25/2020-04-27_14-17-23-right_repeater.mp4'
Mon 27 Apr 19:57:02 EDT 2020: Creating output directory 'SentryClips/2020-04-27_14-17-25'
Mon 27 Apr 19:57:02 EDT 2020: Moving 'SentryClips/2020-04-27_14-17-25/2020-04-27_14-17-23-back.mp4'
Mon 27 Apr 19:57:04 EDT 2020: Moved 'SentryClips/2020-04-27_14-17-25/2020-04-27_14-17-23-back.mp4'
Mon 27 Apr 19:57:04 EDT 2020: Creating output directory 'SentryClips/2020-04-27_14-17-25'
Mon 27 Apr 19:57:04 EDT 2020: Moving 'SentryClips/2020-04-27_14-17-25/event.json'
Mon 27 Apr 19:57:05 EDT 2020: Moved 'SentryClips/2020-04-27_14-17-25/event.json'
Mon 27 Apr 19:57:05 EDT 2020: Creating output directory 'SentryClips/2020-04-27_14-17-25'
Mon 27 Apr 19:57:05 EDT 2020: Moving 'SentryClips/2020-04-27_14-17-25/thumb.png'
Mon 27 Apr 19:57:05 EDT 2020: Moved 'SentryClips/2020-04-27_14-17-25/thumb.png'
Mon 27 Apr 19:57:05 EDT 2020: Creating output directory 'SentryClips/2020-04-27_15-35-33'
Mon 27 Apr 19:57:05 EDT 2020: Moving 'SentryClips/2020-04-27_15-35-33/2020-04-27_15-24-43-front.mp4'

marcone, 2020-04-28:
> Why is archiving stopping in latest build?
> 
> Archivelog snippet since parking at home. I am connected to Pi over SSH as the car and therefore the Pi are both active. Still only few files got copied over from 248.

I'm not sure what you mean by "connected to Pi over SSH as the car", but logging in to the Pi does not keep the car awake, so when the car goes to sleep and powers down the USB ports, your Pi will lose power unless you're powering it via a battery or other power source.

cleanev, 2020-04-28:
Car is on as I am working on some odds and ends and still on as I write this as i am in the car trying to setup few favorites in browser.

What I cannot understand is why did it stop archiving even though the car is on. It was blinking fast for first few files it copied over and currently double blinking that suggests that it has WiFi connection, unless I am wrong

marcone, 2020-04-28:
please attach diagnostics. See https://github.com/marcone/teslausb/issues/35

cleanev, 2020-04-28:
May log seems to be too big so pastebin did not allow me to paste. So I created a Boxnote on my account and here’s the link to that file - https://bimal.box.com/s/1bsc1c663j9gq8el5xww1jh8o8kwv0t2

marcone, 2020-04-28:
It seems the Pi has trouble communicating with your server:
```
[Mon Apr 27 20:06:51 2020] CIFS VFS: Error -11 sending data on socket to server
[Mon Apr 27 20:07:33 2020] CIFS VFS: Error -104 sending data on socket to server
```
I don't know what that means, but it's unrelated to teslausb and I can't really help with that.

closed 2020-04-28 by marcone