marcone/teslausb #477: crossed out dashcam icon

nebukadnezar, 2020-04-29:
I have two pi4's with 256GB sandisks in use. camdisk.bin is 240GB in size. They worked reliably until the snapshots started to loop (camdisk.bin filling up). Data never gets archived, I'm just pulling files of interest via the smb mount, and depend on TeslaUSB deleting old data on its own.

Whenever the dash icon becomes crossed out, I'm resetting/power cycling the Pi4's, and that brings them back online for about an hour, and then they fail again.

It is my understanding that TeslaUSB should automatically delete snapshots when space becomes low, but perhaps it's cutting things too close to actually available space? Or, because the pi4s never have internet connection the date/time never gets updated and that's at the root of the problem? 

/mutable/archiveloop.log doesn't show any errors, only that it is mounting snapshots, and occasionally deleting snapshots due to low space. So that seems to work on the surface.

Any suggestions on how to further debug this?

marcone, 2020-04-29:
> They worked reliably until the snapshots started to loop (camdisk.bin filling up).

I don't understand what you mean by that. cam_disk.bin is the disk image that's presented to the car as a USB drive. The only way it could fill up is if the car fills it up with recordings, but that has nothing to do with snapshots.
How many snapshots do you actually have? With a 256GB card and 240GB cam_disk, I'd expect there to be at most one snapshot.

> Or, because the pi4s never have internet connection the date/time never gets updated and that's at the root of the problem?

If you're not archiving (i.e. you chose archive method "none"), then the time shouldn't matter.

> Any suggestions on how to further debug this?

Have you tried plugging the Pi in to a PC to see if the "CAM" drive shows up and is usable?

nebukadnezar, 2020-04-29:
sorry, typo: The camdisk.bin file is 24GB in size. There are 77 snapshots. And yes, I see both CAM and MUSIC disks when plugging it into a PC. I have noticed the Pi gets very hot though, perhaps there's a temperature problem. I'll keep it in a better ventilated section and see if that makes a difference.

marcone, 2020-04-29:
Pi4 definitely gets a lot warmer than Pi0. Might want to consider one of those heatsink-cases for it.
When you plugged it in to the PC, did you try open the "CAM" drive? Did it still have the TeslaCam folder? Could you write to it?

If it happens again in the car, try getting [diagnostics](https://github.com/marcone/teslausb/issues/35) while it's in that state.

nebukadnezar, 2020-05-01:
This seems to have been a heat related problem. Moving them to a location with better airflow appears to have fixed it.

marcone, 2020-05-01:
Good to hear

closed 2020-05-01 by marcone