marcone/teslausb #479: (Workaround included) /etc/auto.master.d/teslausb.autofs: No such file or directory

maxhac03, 2020-04-29:
Issue:
After updating TeslaUSB to the latest version i got this error message:

> Downloaded /root/bin/auto.teslausb ...
> /tmp/configure-automount.sh: line 35: /etc/auto.master.d/teslausb.autofs: No such file or directory

I did a clean install using the teslausb-20190615 image and got the same error again.

Workaround:
I created both the auto.master.d folder and the teslausb.autofs file and the installation could complete.

marcone, 2020-04-30:
Thanks, looks like perhaps that directory doesn't yet exist immediately after installing autofs. I'll update the setup script to create it.

marcone, 2020-04-30:
Did you only try this with the teslausb-20190615 image?
I can't actually reproduce it with the latest image.

closed 2020-05-01 by marcone