marcone/teslausb #480: Is it possible to have write permissions via Samba?

Jughead135, 2020-04-30:
Just what the title says.

I found the file /etc/samba/smb.conf and tried editing `read only = yes` into `read only = no` and/or `writeable = yes` -- but none of it works.  Moreover, it appears that each reboot of the pi resets the variable to read-only.

Am I in the right place to make this change, or is this just a copy of the real conf file?
Is it simply incompatible with the teslausb architecture?
Is it a "really bad idea" to do any file management vis Samba?

I have some random files that have been copied to random places, and if I can ID them, I figure why not try to put them where they belong.  Doing so via my Win10 PC would be about a billion times easier than using the ssh command line (I'm not even sure how that would be done, anyway).  Windows dutifully informs my any time I try that I don't have sufficient permissions....

marcone, 2020-04-30:
It's not clear to me what you're actually trying to do. The files that you see when accessing the Pi via Samba are inside the snapshots, and are neither visible to the car nor archived.

Jughead135, 2020-04-30:
Understood.  Sorry I as not more clear.

I'm looking at the various files in the RecentClips folder ("folder" as viewed in Win10 over Samba).  Among the folders with the familiar date naming convention, there are a number of folders with weird names--some have a single vid file (generally also with a weird name), others will have an otherwise normal looking collection of vids.  Most of those vids are generally identifiable via timestamp, so moving them to the folder where they're missing would be trivial...  if I had write permissions.

This is probably something I'll stop caring about once I've gotten used to how the dashcam & teslausb work together.  Really, how often will I be viewing dashcam footage??  I assume the random files/folders will drop off over time as the storage fills up along with all of the rest of the older files, so, there's no "reason" to have to do it.  More about me learning-by-doing (and pestering the adults along the way).  Now that I've got the Samba set up (thanks again to your pointers), I was just exploring its limits vis-a-vis the teslasusb.

marcone, 2020-04-30:
Oh, you mean names like "202FFA~2.MP4" and similarly "garbage" names?
Those are an artefact of the interaction between the car and teslausb. When teslausb takes a snapshot of the active drive image, it runs a filesystem check on that snapshot. That filesystem check may then recover a deleted file, or a file that hasn't been finalized yet. It then generally assigns the recovered file a bogus name. A similar thing may happen with folders.
The important thing to note here is that these files will be either incomplete, or complete copies of files that already exist in another snapshot, so there's no need to go and rename them or to copy them to their proper location. They should probably just be deleted but I haven't gotten around to that.

For example, I have the following files in my current collection of snapshots:
```
pi@teslausb:/backingfiles/TeslaCam $ ls -lL $(find -type l -name \*\~\*)
-rwxrwxrwx 1 root root 28892263 Apr 23 12:37 ./RecentClips/200A1D~3.M/200A1D~3.MP4
-rwxrwxrwx 1 root root 29532665 Apr 26 08:55 ./RecentClips/202FFA~2.M/202FFA~2.MP4
-rwxrwxrwx 1 root root 27056817 Apr 18 08:46 ./RecentClips/20764C~7.M/20764C~7.MP4
```
(I'm using the "-L" option above because /backingfiles/TeslaCam, which is the directory that you see via Samba, contains symlinks to the actual files in the snapshots, and I want to follow the symlinks)
If I go and look for files with the same sizes as these 3, I get:
```
pi@teslausb:/backingfiles/TeslaCam $ find /tmp/snapshots/ -size 28892263c -o -size 29532665c -o -size 27056817c
/tmp/snapshots/snap-000020/TeslaCam/RecentClips/202FFA~2.MP4
/tmp/snapshots/snap-000021/TeslaCam/RecentClips/2020-04-26_15-54-09-front.mp4
/tmp/snapshots/snap-000013/TeslaCam/RecentClips/200A1D~3.MP4
/tmp/snapshots/snap-000014/TeslaCam/RecentClips/2020-04-23_19-36-54-front.mp4
/tmp/snapshots/snap-000008/TeslaCam/RecentClips/2020-04-18_15-45-35-front.mp4
/tmp/snapshots/snap-000007/TeslaCam/RecentClips/20764C~7.MP4
```
and I can verify that these are actually duplicates by calculating their checksums:
```
pi@teslausb:/backingfiles/TeslaCam $ md5sum $(find /tmp/snapshots/ -size 28892263c -o -size 29532665c -o -size 27056817c)
236a6a1151b8d2557b51b5c7d6e793f9  /tmp/snapshots/snap-000020/TeslaCam/RecentClips/202FFA~2.MP4
236a6a1151b8d2557b51b5c7d6e793f9  /tmp/snapshots/snap-000021/TeslaCam/RecentClips/2020-04-26_15-54-09-front.mp4
dab239ef9fc4cd13e808a1df687125eb  /tmp/snapshots/snap-000013/TeslaCam/RecentClips/200A1D~3.MP4
dab239ef9fc4cd13e808a1df687125eb  /tmp/snapshots/snap-000014/TeslaCam/RecentClips/2020-04-23_19-36-54-front.mp4
ebdac537edfb9a3adb12ec9a61ab2d66  /tmp/snapshots/snap-000008/TeslaCam/RecentClips/2020-04-18_15-45-35-front.mp4
ebdac537edfb9a3adb12ec9a61ab2d66  /tmp/snapshots/snap-000007/TeslaCam/RecentClips/20764C~7.MP4
```
I don't really understand why the bogus name is in the earliest snapshot in all three cases. I would have expected it to be in the later one, after having been deleted by the car and then recovered by the filesystem check.  ¯\_(ツ)_/¯
The important thing to note though is that there exist properly-named copies of the bogus files, and so you can just ignore the bogus ones.

Jughead135, 2020-04-30:
> The important thing to note though is that there exist properly-named copies of the bogus files, and so you can just ignore the bogus ones.

OK, cool--I didn't realize that they'd already been copied over.  Thanks for the detailed walk-through, I'm enjoying each opportunity to peek under the hood here.

So, even though this specific "reason" is not really a reason after all:  how do I give myself write privileges through Samba?  The conf file I found apparently ain't it.

If Samba has write privs, would making any changes through the Win10 GUI cause Really Bad Things to happen, vice making those changes via the SSH command line?

marcone, 2020-04-30:
Samba gives you read-only access to directories with symlinks to files that are themselves located in snapshots that are mounted read-only. Even if you made both writeable, your next problem then is that Samba is set up to follow those symlinks, so if you tried to rename a file for example, you would rename the target file, leaving the newly renamed target file inaccessible because there is no longer a symlink pointing to it, while the old symlink would be stale because it no longer points to anything. Or maybe it would rename the link but not the file, I haven't actually tried it. Either way, it's not what I would call Really Bad (it wouldn't really break anything), but it's certainly not what you'd want either.

Jughead135, 2020-04-30:
> Either way, it's not what I would call Really Bad (it wouldn't really break anything), but it's certainly not what you'd want either.

You're correct, as usual--not what I'm trying to do.  Obviously I'm not up to speed on Samba, since that was a stupid question in light of your description of how it operates.  Thanks again.

marcone, 2020-04-30:
No, not a stupid question at all. Teslausb is using a bit of a nonstandard Samba setup. Normally you wouldn't want a Samba server to follow symlinks, because it can be a security issue: a user could create symlinks to files outside the file share, and then access the files they point to with Samba's privileges. For teslausb's handling of snapshots though, it was required.

closed 2020-05-01 by marcone