marcone/teslausb #481: Rclone erroring with SSL certificate

handshake87, 2020-05-06:
Following you guide advises to use curl to install rclone. While following these steps it actually never install with an error stating SSL certificate is missing. Any solution for this? Maybe add this step in the rclone guide

marcone, 2020-05-06:
What's the actual error you're seeing?

handshake87, 2020-05-06:
I get the following error message:

curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: http://curl.haxx.se/docs/sslcerts.html

marcone, 2020-05-06:
and what command are you running to get that error?

When I run `curl https://rclone.org/install.sh | sudo bash`, which is the command listed on [teslausb's rclone setup page](https://github.com/marcone/teslausb/blob/main-dev/doc/SetupRClone.md), and is a direct copy from the [rclone website](https://rclone.org/downloads/), it works just fine.

thijsdejong, 2020-05-06:
I actually got this same error when trying to install a freshly generated image. It was unable to because of that exact ssl error message.

marcone, 2020-05-06:
By "freshly generated" do you mean you built your own image?

marcone, 2020-05-06:
Also, is it only the curl for rclone that fails, or does teslausb itself fail to install? (since that also uses curl pull the scripts from github)

handshake87, 2020-05-06:
Only the rcurl command. While trying to grab it we get a SSL error.

thijsdejong, 2020-05-07:
> By "freshly generated" do you mean you built your own image?

Yes I built my own image. It's curl that fails in general, so headless install fails aswell. Thinking it might be related.

sebnaf, 2020-05-14:
Please check your system date/time. In my case the Pi was way off and all SSL certs failed validation (yes, still ntpd issues…)

marcone, 2020-08-12:
Closing because this issue over 3 months old, and I can't reproduce it

closed 2020-08-12 by marcone