marcone/teslausb #482: Feature request: Watch filesystem for new files, instantly copy to share

rmariboe, 2020-05-12:
I'm building a setup where the Pi is constantly online (either via LTE dongle or LTE connected wifi router in car) and connected to NAS through OpenVPN - thus, the Pi never leaves/reaches wifi connectivity and therefore, I assume nothing will trigger archiving.

Request is to add a filesystem watcher for files that are created, triggering a copy when they are closed. Is this viable?

marcone, 2020-05-12:
There is no filesystem for the Pi to watch: the car writes the files to a filesystem in a disk image, which isn't mounted on the Pi while it is accessed by the car.
In order to do what you describe, teslausb would have to look at the content of that disk image at regular intervals, which is technically possible, and something I hope to get to eventually, but it wouldn't be as instant as you seem to be asking for.

marcone, 2020-05-12:
Closing this in favor of the older issue https://github.com/marcone/teslausb/issues/171 which is asking for the same thing.

closed 2020-05-12 by marcone

rmariboe, 2020-05-12:
Ah, and after spending a few hours looking for a good way to monitor the filesystem being written through the USB mount, I'm about to give up on that path. Guess I'll just force_sync once every few minutes or so...

marcone, 2020-05-12:
teslausb disconnects the drive from the car while archiving (because you shouldn't have two different computers accessing the same disk at the same time), so you would be interrupting any sentry recordings in progress.

rmariboe, 2020-05-13:
Damned, feared as much. So I'll have to create a custom script mounting the cam_disk and the archive and copying the files I guess - wonder what'll happen if the car starts deleting files while they are being copied :)

amirasn, 2021-04-15:
Hi @rmariboe . Did you succeed in implementing the solution for Pi to be constantly online ?

scifiguy000, 2024-08-25:
I also have had situations where I would like to review Sentry clips remotely.  On one occasion, my car was hit in a parking lot.  I got a Sentry notification on my phone, but when I looked through the camera view, the driver had already left.  I only found out much later when I went to my car that it had been hit.  If I had a way to review the event that triggered the alert, I would have known to call police immediately.  It worked out eventually, because the police tracked the hit and run driver by license plate from my Sentry videos, and the other party confessed (and, luckily, they were insured).

I'm using Rclone and a cellular hotspot.  Would periodically executing "touch /tmp/archive_is_unreachable" trigger Sentry clip uploads (in this case, Google Drive)?  I assume the car would miss any Sentry events happening while archiving.

amirasn, 2024-08-28:
@scifiguy000 Interesting use case! We have developed a product that does that and also comes with an app that you can track your trips and make video request based the trips you make: https://discord.com/channels/629413353223225355/629413353223225357/1268029806440747110