marcone/teslausb #484: TeslaUSB shutdown the PIZero

mrteausa, 2020-05-12:
I have 2018 MX with MCU3 computer.  First get in the car, the TeslaUSB it took about 10 seconds to connect and for the USB music and Cam to show up.  How can the Sentry mode turn on it it cannot detect the TeslaCAM folder when it timeout and shutdown?  Can I connect the PIZero power plug to the 12v constant from the car directly to the battery and the middle USB data to the car USB data?  Will it work and keep the TeslaUSB drive to show up and PiZero to stay on all the time?

marcone, 2020-05-12:
Yes, you can power the Pi0 from its "power" port while connecting the "USB" port to the car. If you do that, you should make sure you're using a "data only" cable for the USB side, [as noted here](https://github.com/marcone/teslausb/wiki/Troubleshooting).

That being said, the car's USB ports are powered while Sentry mode is on, so it's not really necessary to have a separate power supply.

closed 2020-05-12 by marcone

mrteausa, 2020-05-12:
I also noticed that when the power is on and connected to the WIFI always, in garage when I copy some music files to my share TeslaMusic folder it does not update the TeslaUSB Music folder.  It only sync the folder when I reboot and restart the PIZero.  Is there a fix to check for files changes regularly?

marcone, 2020-05-12:
There isn't. Teslausb only archives recordings [at specific times](https://github.com/marcone/teslausb/wiki/FAQ#q-i-manually-copied-some-files-to-the-cam-drive-but-theyre-not-being-archived), and music is synced at the same time.