marcone/teslausb #489: CAM share doesn't appear

manosinistra, 2020-05-29:
I'm using the latest version on a fresh install.

I cannot get the CAM share to appear on my computer. My Tesla doesn't see the drive either.

However, if I run the following manually as root:

/sbin/modprobe -r g_mass_storage
/sbin/modprobe g_mass_storage

then the share appears on my computer and in-car the dashcam appears.

Any ideas?

marcone, 2020-05-29:
Please see https://github.com/marcone/teslausb/issues/35

manosinistra, 2020-05-29:
Looks like I'm having more problems than originally thought.

I just did a complete reinstall following your pre-built image instructions on a Pi0w.

Long story short, it looks like rc.local never runs (this is fresh install). I had already updated the setup and wpa .conf files.

Couldn't figure out why rc.local doesn't run; I scoured the web at length. Ultimately had to add a line in the root crontab:

`@reboot /bin/bash /etc/rc.local`

Then I had to run the setup again.

And still I didn't get the CAM share to show.

Had to add the 2 modprobe lines to /etc/rc.local.

Now it all works...

marcone, 2020-05-29:
None of that should be necessary when using the prebuilt image.
rc.local is run by a systemd service, not by cron, and the archiveloop script runs modprobe as needed, to avoid mounting the disk image on the car and the Pi at the same time.

closed 2020-05-29 by marcone

manosinistra, 2020-05-29:
Hi,

Yes, I get that. Just letting you know that's what happened. Clean install
of the image using BalenaEtcher. Not sure what the issue could be.

Also, without the modprobe the USB won't mount to the computer(s) or Tesla.

Anyways, thanks for everything that you do for us.

Thanks,

Danny

marcone, 2020-05-29:
And that's why I directed you towards issue https://github.com/marcone/teslausb/issues/35, which links to troubleshooting steps, and mentions `setup-teslausb diagnose`, which probably would have clarified the problem.

marcone, 2020-05-29:
(and to be clear: by adding those extra calls to modprobe, you may end up exposing the drive image to the car and the Pi at the same time, which will almost certainly lead to corruption)