marcone/teslausb #491: Trying to get ssh pi@telsausb.local to work

bit-bash, 2020-06-03:
My Pi Zero is working with teslausb, and I'd like to ssh to it from my Ubuntu 18.04 PC via the USB connection to the Pi Zero (the Pi is connected to my PC on its USB data port, NOT the power port).

The hostname of my Pi Zero is: teslausb
It's up and running with a 1 blink pattern and my linux PC can see the mounted MUSIC and CAM directories OK.

I can connect to it via wifi using ssh pi@192.168.1.20 but I'd like to make the connection via the USB networking (assuming this is supposed to work).  Do I need to do something in Ubuntu to get it to connect to the Pi Zero via USB networking so that pi@teslausb.local works?

Thanks!

marcone, 2020-06-03:
This is not supported. It would require using the "g_multi" module, which combines the mass storage gadget, ethernet gadget and usb serial gadget into one. The problem then is that as soon as teslausb starts archiving, you would lose the network connection as the module would be unloaded.

I tried getting around that by using configfs instead of the g_mass_storage module, since that would allow usb networking to operate independently of mass storage, however it seemed that the configfs version of the mass storage gadget isn't as stable as the g_mass_storage module, because I kept getting errors after hours or days of usage.

closed 2020-06-03 by marcone

bit-bash, 2020-06-03:
Ahh, no problem.  If it did work it would be handy (wouldn't have to figure out what IP address the Pi was at).  But using the WiFi is fine!
Thanks!

marcone, 2020-06-03:
You shouldn't normally have to figure out the IP address for the Pi, "teslausb.local" should work on most systems, including Ubuntu 18.04 (at least it works on my 18.04.4 LTS machine)

bit-bash, 2020-06-03:
Ya, ssh pi@teslausb.local doesn't work for me.  I suspect that my router needs to support mDNS (old zeroconfig) which it doesn't - it's old.  ;)

marcone, 2020-06-03:
I'm no expert (I pretty much just know that "avahi" is involved on Ubuntu), but I thought the whole point of Bonjour/zeroconf was that it works in existing networks and doesn't need any special configuration of routers or DNS servers.

bit-bash, 2020-06-03:
Gosh, I'm not expert either - just guessing along those lines.  My ubuntu does have avahi up and running but now I wonder if my issue is that my WiFi is an access point on my network - the WiFi is NOT part of my router that hands out IP addresses to the local systems (it's a separate box) - so maybe that prevents avahi from "discovering" telsausb.local.  Well, at least I can connect to it via its IP address.  ;)