marcone/teslausb #493: how does the car and the raspberry pi write to the filesystem at the same time?

AndySchroder, 2020-06-09:
No where in the documentation does it describe how both the car and the raspberry pi write to the filesystem at the same time. Could you please explain this in the readme file?

marcone, 2020-06-09:
It's not described in the documentation because they don't.

closed 2020-06-09 by marcone

AndySchroder, 2020-06-09:
Okay, then how do video files get removed from the teslausb raspberry pi computer if the car is writing to the disk? How are music files written if the car is reading from the disk? It doesn't make any sense.

marcone, 2020-06-09:
At any given time, the drives are visible either to the Pi or the car. Never to both at the same time. Everything follows from that: the only videos that get removed while the car is recording are the videos that the car itself is removing because they're older than 1 hour. When music is being copied, the car doesn't see the music drive, so the car won't be reading from it. When teslausb is archiving, the car doesn't see the cam drive, so the car won't be writing to it.

AndySchroder, 2020-06-10:
Very cool! So the software is able to virtually yank the USB disk from the USB port in the car? How does this work out if well you are constantly improperly unmounting the disk from the car?

marcone, 2020-06-10:
It doesn't do it "constantly". Why would it?
You might want to have a look at [the main loop in the code](https://github.com/marcone/teslausb/blob/main-dev/run/archiveloop#L508). It's fairly straightforward and might help you understand what exactly teslausb does.