marcone/teslausb #495: new kernel breaks install: disk full

hb020, 2020-06-11:
### Configuration

Latest teslausb image (teslausb-20200122)
Fresh SD card (128GB)

### Problem description

When installing from the latest teslausb image, it fails after reading configure-automount.sh.

From the progress log:

```
....
Downloaded /tmp/configure-automount.sh ...
W: --force-yes is deprecated, use one of the options starting with --allow instead.
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem.
```

Upon research, it is `apt upgrade` fails with that last message.
So when running `dpkg --configure -a`, followed by `apt upgrade`, I saw that the install of the new kernel 20200601 fails due to:

```
Unpacking raspberrypi-kernel (1.20200601-1) over (1.20200114-1) ...
dpkg: error processing archive /var/cache/apt/archives/raspberrypi-kernel_1.20200601-1_armhf.deb (--unpack):
 cannot copy extracted data for './lib/modules/4.19.118-v8+/kernel/drivers/block/zram/zram.ko' to '/lib/modules/4.19.118-v8+/kernel/drivers/block/zram/zram.ko.dpkg-new': failed to write (No space left on device)
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
```

Indeed, `/dev/root` has 86MB free. Time for a bigger disk image, or putting `/var` elsewhere?

### Workaround
```
apt upgrade
apt clean
apt upgrade
```
apt upgrade will have succeeded.
After that,
`/etc/rc.local` will now fail with:
```
Downloaded /tmp/configure.sh ...
~ /
Fixing the modules-load parameter in /boot/cmdline.txt...
Fixed cmdline.txt.
USB_DRIVE not set. Proceeding to SD card setup
create-backingfiles-partition: using existing backingfiles and mutable partitions
create-backingfiles-partition: using existing backingfiles and mutable partitions
TARGET        SOURCE         FSTYPE OPTIONS
/backingfiles /dev/mmcblk0p3 xfs    rw,noatime,attr2,inode64,noquota
umount: /mnt/cam: not mounted.
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/4.19.93+/modules.dep.bin'
STOP: Couldn't check image
```
That is normal, as that refers to an old version that is no longer there.

Then reboot, and the system comes up and finishes the install (after 10+minutes).

marcone, 2020-06-13:
Why did it run `apt upgrade` in the first place? That's not a command that teslausb setup runs.
(it might run `apt-get upgrade`, but only if the UPGRADE_PACKAGES environment variable is true, and that's not part of the sample config)

hb020, 2020-06-13:
Maybe I indeed set UPGRADE_PACKAGES (don't remember anymore), maybe I was falsely lead to believe it did an upgrade, as the setup script (without me interfering before hand other than preparing the config file) complained about having to manually run `sudo dpkg --configure -a`. And that is an error one normally gets when doing an upgrade after a disk problem.
Anyway, something went wrong, and at least the present default disk size config it will not survive an upgrade automatically. A too small predefined disk is part of the issue. Not saying that it is the only cause, nor that increasing it is the only solution.
Really sorry to drop this off here and not being able to dig in deeper for now, as my priorities have taken a radical change these last days. Just thought that you might want to look at this to avoid problems, since the upgrades will only get bigger over time.

marcone, 2020-06-13:
I'm seeing about 440 MB free on the root partition when doing a standard install using the latest image, once installation is finished.
After package upgrade, it has between 240 and 290 MB free (depending on whether I run it as part of setup or manually afterwards), though it does drop down to 0 MB and into the reserved space during the upgrade (I think mainly because it will temporarily have two kernels and their modules installed, before deleting the old one). 

It's clear something went wrong in your case (my best guess is a brief power drop that interrupted dpkg and left things in an inconsistent state), but without the setup log it's impossible to say what went wrong, or what could have been done about it. Since there's no way to handle every edge case, and there doesn't appear to be an  inherent problem with the latest image or setup, I'm going to close this.

closed 2020-06-13 by marcone