marcone/teslausb #496: How to update the TeslaUSB file share settings and nothing else?

miles267, 2020-06-14:
TeslaUSB has been running great for a long time.  Flawless.  However I recently built a new file server and moved my SMB file share to it.

I only want to update TeslaUSB to point to that new location without re-running the entire re-config on TeslaUSB.

Is this possible?  Thank you.

marcone, 2020-06-14:
If you truly don't want to update anything else, you can manually update /etc/fstab with the new share info. You will also need to update /root/teslausb.conf (if that doesn't exist then update /root/teslausb_setup_variables.conf instead), and if you changed the user/password for the share, update /root/.teslaCamArchiveCredentials

closed 2020-06-14 by marcone

miles267, 2020-06-14:
Thank you @marcone. This works exactly as I had hoped. TeslaUSB immediately began transferring the queued footage upon reconnect to new SMB file share.