marcone/teslausb #497: Pi Zero W not connecting to WiFi anymore

joejoethejoey, 2020-06-15:
After the lastest Tesla Update (2020.20.12) the car wouldn't detect the TeslaUSB.

I've tried to flash a fresh copy using my previously saved conf file, but now it doesn't seem to connect to my WiFi network anymore.  Could the WiFi chip have been fried during an update?  It still powers on and I see it load.  But can't SSh into it or even ping it.

marcone, 2020-06-15:
When you say "I see it load", what do you mean?
Does the LED on the Pi blink regularly, once or twice per second?
When you plug the Pi in to a PC, do the CAM and/or MUSIC drives show up after a little while?

joejoethejoey, 2020-06-15:
Thanks marcone.  Finally managed to work out it was my 2 old SDCARDs.  I tried a new one today and it is working fine now.  It must be corrupted somehow.  All good!!!  Cheers!

closed 2020-06-15 by joejoethejoey