marcone/teslausb #498: FritzBox blocks CIFS

TheLastProject, 2020-06-21:
A FritzBox router will silently drop CIFS unless the NetBIOS filter is disabled under `Internet` -> `Filters` -> `Lists`. It will even do so if you don't actually use NetBIOS, the filter just forbids all incoming traffic to port 139 and 445. This may be worth mentioning in troubleshooting for issues related to "CIFS not working" and it's a fairly popular router in the Netherlands (provided by the Dutch ISPs Freedom Internet and XS4ALL) and from what I know Germany too.

marcone, 2020-06-21:
If I'm reading the Fritzbox documentation correctly, that packet filter only applies to internet->local traffic, not for local->local traffic. In other words, it will prevent someone on the internet from trying to access your local network's CIFS share (which you definitely want to prevent!), but shouldn't block a computer on your local network from accessing the CIFS server that is also on your local network.

The normal use case is for the Pi to be on your local network, so traffic from the Pi to the CIFS share would bypass the router's packet filter. That would be no different than connecting to your CIFS share from e.g. your laptop, which I assume works, and doesn't require you to go and disable the packet filter?

TheLastProject, 2020-06-21:
Ah, yeah, then I suppose it doesn't affect most people at all. In my case, my Wi-Fi can't possibly reach the building's shared garage so I actually create a hotspot with my phone for the Pi to use instead.

I guess I'll close this as this will probably be too uncommon to be worth a troubleshooting entry and at least there's this GitHub issue now which may show up on a search.

closed 2020-06-21 by TheLastProject