marcone/teslausb #500: Snapshots not working

maxhac03, 2020-06-29:
Since the implementation of Automount, TeslaUSB is no longer making new snapshots correctly for me.

```
Starting archiveloop at 13.45 seconds uptime...
taking snapshot of cam disk in /backingfiles/snapshots/snap-000002
took snapshot
comparing new snapshot with /backingfiles/snapshots/snap-000001/snap.bin
new snapshot is identical to previous one, discarding
```
Its not identical as snap-000002 does contain the latest videos from RecentClips after the snapshot process. (So it is not discarded)

If i comment the section that verify the differences between the new and previous snapshot in make_snapshot.sh i get this:
```
Starting archiveloop at 13.61 seconds uptime...
taking snapshot of cam disk in /backingfiles/snapshots/snap-000002
took snapshot
```

In both cases, snap-000002 contain the latest videos from RecentClips.

TeslaUSB is always overwriting snap-000002. I am using a 512GB Micro SD card on a Pi ZeroW.

Tried multiple times to do a fresh install with a fresh teslausb_setup_variables.conf.

[teslausb_setup_variables.conf.txt](https://github.com/marcone/teslausb/files/4847201/teslausb_setup_variables.conf.txt)

marcone, 2020-06-29:
What exactly did you change to make it keep the snapshot, and is snap-000002 usable after that?
The comparison between snapshots is pretty straightforward, so I don't see why it could be thinking that the snapshots are the same when they're not.

maxhac03, 2020-06-29:
I did this:
```
  # check whether this snapshot is actually different from the previous one
#  find "$newsnapmnt/TeslaCam" -type f -printf '%s %P\n' > "${newsnapname}.toc_"
#  log "comparing new snapshot with $oldname"
#  echo '>dummy' > "${newsnapname}.toc_"
#  if [[ ! -e "${oldname}.toc" ]] || diff "${oldname}.toc" "${newsnapname}.toc_" | grep -e '^>'
#  then
#    ln -s "$newsnapmnt" "$newsnapdir/mnt"
#    make_links_for_snapshot "$newsnapmnt" "$newsnapdir/mnt"
#    mv "${newsnapname}.toc_" "${newsnapname}.toc"
#  else
#    log "new snapshot is identical to previous one, discarding"
#    /root/bin/release_snapshot.sh "$newsnapdir"
#    rm -rf "$newsnapdir"
#  fi
```

However in both cases snap-000002 still contain the latest video files.The snapshots are always made as snap-000002.

marcone, 2020-06-29:
That's going to skip the crucial steps of generating the table of contents for the new snapshots, moving it to the correct location, creating the mountpoint link and generating the symlinks for Samba access.
Try instead just replacing:
```
  if [[ ! -e "${oldname}.toc" ]] || diff "${oldname}.toc" "${newsnapname}.toc_" | grep -e '^>'
```
with
```
  if true
```
Then afterwards you can manually run `bin/make_snapshot.sh` and compare snap-000001/snap.bin.toc and snap-000002/snap.bin.toc

maxhac03, 2020-06-29:
I will do a clean install of teslausb-20200122 and i will do this.
I will enable Sentry mode to keep the car awake in my garage and report the results.

Thanks.

maxhac03, 2020-06-29:
```
  # check whether this snapshot is actually different from the previous one
  find "$newsnapmnt/TeslaCam" -type f -printf '%s %P\n' > "${newsnapname}.toc_"
  log "comparing new snapshot with $oldname"
#  if [[ ! -e "${oldname}.toc" ]] || diff "${oldname}.toc" "${newsnapname}.toc_" | grep -e '^>'
   if true
  then
    ln -s "$newsnapmnt" "$newsnapdir/mnt"
    make_links_for_snapshot "$newsnapmnt" "$newsnapdir/mnt"
    mv "${newsnapname}.toc_" "${newsnapname}.toc"
  else
    log "new snapshot is identical to previous one, discarding"
    /root/bin/release_snapshot.sh "$newsnapdir"
    rm -rf "$newsnapdir"
  fi
```
```
root@TeslaUSB-Model3:~# /root/bin/make_snapshot.sh
/root/bin/make_snapshot.sh: line 146: log: command not found
/backingfiles/snapshots/snap-000002/snap.bin on /tmp/snapshots/snap-000002 type vfat (ro,nosuid,nodev,noexec,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro)
snapshot already mounted
fsck from util-linux 2.33.1
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Performing changes.
/dev/loop0p1: 25 files, 93663/2094976 clusters
/root/bin/make_snapshot.sh: line 148: log: command not found
/root/bin/make_snapshot.sh: line 152: log: command not found
ln: failed to create symbolic link '/backingfiles/snapshots/snap-000002/mnt/snap-000002': Read-only file system
/root/bin/make_snapshot.sh: line 53: log: command not found
/root/bin/make_snapshot.sh: line 89: log: command not found
```
```
root@TeslaUSB-Model3:~# cat /backingfiles/snapshots/snap-000001/snap.bin.toc
root@TeslaUSB-Model3:~# cat /backingfiles/snapshots/snap-000002/snap.bin.toc
38112281 RecentClips/2020-06-29_17-51-06-front.mp4
38031589 RecentClips/2020-06-29_17-51-06-left_repeater.mp4
37951772 RecentClips/2020-06-29_17-51-06-right_repeater.mp4
39227170 RecentClips/2020-06-29_17-51-06-back.mp4
37892932 RecentClips/2020-06-29_17-52-07-front.mp4
38188307 RecentClips/2020-06-29_17-52-07-left_repeater.mp4
38118021 RecentClips/2020-06-29_17-52-07-right_repeater.mp4
39216463 RecentClips/2020-06-29_17-52-07-back.mp4
38001438 RecentClips/2020-06-29_17-55-27-front.mp4
37966742 RecentClips/2020-06-29_17-55-27-left_repeater.mp4
38212596 RecentClips/2020-06-29_17-55-27-right_repeater.mp4
39146121 RecentClips/2020-06-29_17-55-27-back.mp4
38158440 RecentClips/2020-06-29_17-56-28-front.mp4
38143351 RecentClips/2020-06-29_17-56-28-left_repeater.mp4
37997079 RecentClips/2020-06-29_17-56-28-right_repeater.mp4
39131083 RecentClips/2020-06-29_17-56-28-back.mp4
37933552 RecentClips/2020-06-29_17-57-28-front.mp4
38244156 RecentClips/2020-06-29_17-57-28-left_repeater.mp4
38180731 RecentClips/2020-06-29_17-57-28-right_repeater.mp4
39317968 RecentClips/2020-06-29_17-57-28-back.mp4
38158532 RecentClips/2020-06-29_17-58-29-front.mp4
38098488 RecentClips/2020-06-29_17-58-29-left_repeater.mp4
37959669 RecentClips/2020-06-29_17-58-29-right_repeater.mp4
39049243 RecentClips/2020-06-29_17-58-29-back.mp4
37937332 RecentClips/2020-06-29_17-59-29-front.mp4
6160384 RecentClips/2020-06-29_17-59-29-left_repeater.mp4
```

[archiveloop.log](https://github.com/marcone/teslausb/files/4848030/archiveloop.log)

marcone, 2020-06-29:
Can you `rm -rf /backingfiles/snapshots/snap-000002` and try again?

The problem might be that the link to the mountpoint already exists, and it's trying to create it again. At least that's the only explanation I can think of for this failure:
```
ln: failed to create symbolic link '/backingfiles/snapshots/snap-000002/mnt/snap-000002': Read-only file system
```

maxhac03, 2020-06-29:
Removed snap-000002 and rebooted:
```
Mon 29 Jun 18:44:35 ADT 2020: Starting archiveloop at 19.60 seconds uptime...
Mon 29 Jun 18:44:37 ADT 2020: taking snapshot of cam disk in /backingfiles/snapshots/snap-000002
Mon 29 Jun 18:44:42 ADT 2020: took snapshot
Mon 29 Jun 18:44:42 ADT 2020: comparing new snapshot with /backingfiles/snapshots/snap-000001/snap.bin
Mon 29 Jun 18:44:42 ADT 2020: making links for /tmp/snapshots/snap-000002, retargeted to /backingfiles/snapshots/snap-000002/mnt
Mon 29 Jun 18:44:42 ADT 2020: made all links for /tmp/snapshots/snap-000002
```
Removed snap-00002 then executed make_snapshot.sh:
```
root@TeslaUSB-Model3:~# rm -rf /backingfiles/snapshots/snap-000002
root@TeslaUSB-Model3:~# /root/bin/make_snapshot.sh
/root/bin/make_snapshot.sh: line 146: log: command not found
/backingfiles/snapshots/snap-000002/snap.bin (deleted) on /tmp/snapshots/snap-000002 type vfat (ro,nosuid,nodev,noexec,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro)
snapshot already mounted
fsck from util-linux 2.33.1
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Long filename fragment "2020-06-29_18" found outside a LFN sequence.
  (Maybe the start bit is missing on the last fragment)
  Not auto-correcting this.
Performing changes.
/dev/loop1p1: 127 files, 562346/2094976 clusters
/root/bin/make_snapshot.sh: line 148: log: command not found
/root/bin/make_snapshot.sh: line 152: log: command not found
/root/bin/make_snapshot.sh: line 53: log: command not found
/root/bin/make_snapshot.sh: line 89: log: command not found
```
Executed it again:
```
root@TeslaUSB-Model3:~# /root/bin/make_snapshot.sh
/root/bin/make_snapshot.sh: line 146: log: command not found
/backingfiles/snapshots/snap-000002/snap.bin (deleted) on /tmp/snapshots/snap-000002 type vfat (ro,nosuid,nodev,noexec,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro)
snapshot already mounted
fsck from util-linux 2.33.1
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Long filename fragment "2020-06-29_18" found outside a LFN sequence.
  (Maybe the start bit is missing on the last fragment)
  Not auto-correcting this.
Performing changes.
/dev/loop0p1: 131 files, 581042/2094976 clusters
/root/bin/make_snapshot.sh: line 148: log: command not found
/root/bin/make_snapshot.sh: line 152: log: command not found
ln: failed to create symbolic link '/backingfiles/snapshots/snap-000002/mnt/snap-000002': Read-only file system
/root/bin/make_snapshot.sh: line 53: log: command not found
/root/bin/make_snapshot.sh: line 89: log: command not found
```

marcone, 2020-06-29:
it might be useful to run `function log { echo "$@"; }; export -f log` before running make_snapshot, so you don't get all those "log: command not found" messages.

maxhac03, 2020-06-29:
```
root@TeslaUSB-Model3:~# function log { echo "$@"; }; export -f log
root@TeslaUSB-Model3:~# /root/bin/make_snapshot.sh
taking snapshot of cam disk in /backingfiles/snapshots/snap-000002
fsck from util-linux 2.33.1
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Long filename fragment "2020-06-29_18" found outside a LFN sequence.
  (Maybe the start bit is missing on the last fragment)
  Not auto-correcting this.
Performing changes.
/dev/loop0p1: 174 files, 753696/2094976 clusters
took snapshot
comparing new snapshot with /backingfiles/snapshots/snap-000001/snap.bin
ln: failed to create symbolic link '/backingfiles/snapshots/snap-000002/mnt/snap-000002': Read-only file system
making links for /tmp/snapshots/snap-000002, retargeted to /backingfiles/snapshots/snap-000002/mnt
made all links for /tmp/snapshots/snap-000002
```

marcone, 2020-06-30:
This:

> ln: failed to create symbolic link '/backingfiles/snapshots/snap-000002/mnt/snap-000002': Read-only file system

indicates that the symbolic link that it's trying to create already exists. This should not happen, because the "snap-000002" folder is newly created.

aberen001, 2020-07-04:
Pardon me, but what is a snapshot? I saw a new entry in the config file for it, but I'm not sure what it is

marcone, 2020-07-05:
> Pardon me, but what is a snapshot? I saw a new entry in the config file for it, but I'm not sure what it is

They're not exactly new. Snapshots are the mechanism that lets teslausb keep more than an hour of "recentclips". You can access these recordings via Samba.

marven, 2020-07-19:
For some reason on a fresh install my I end up having a folder called `/bacakingfiles/snapshots/snap--00001` , is that intentional?

`oldnum` on this [line](https://github.com/marcone/teslausb/blob/6f67cd83270cbc4f751bedc35fa09bad6d394ab1/run/make_snapshot.sh#L125) ends up being `1` always, so `newnum` ends up being `2` which gets `/bacakingfiles/snapshots/snap-000002` repeatedly overwritten.

marcone, 2020-07-19:
> For some reason on a fresh install I end up having a folder called `/backingfiles/snapshots/snap--00001` , is that intentional?

No, definitely not. The number should be 6 digits, not 5, and there should only be one dash.
Which image are you using for your fresh install?

maxhac03, 2020-07-19:
Removing the snap--00001 seems to be the solution. I will let sentry mode on all night to test.

marven, 2020-07-19:
> > For some reason on a fresh install I end up having a folder called `/backingfiles/snapshots/snap--00001` , is that intentional?
> 
> No, definitely not. The number should be 6 digits, not 5, and there should only be one dash.
> Which image are you using for your fresh install?

The latest one, `teslausb-20200709`.

jun3280net, 2020-07-25:
hi there - it seems like I'm experiencing similar issues with a recent reinstall of teslausb as well.  I have just removed snap--00001 folder and contents as well.  I will report back with my findings.  That said, I was going through my clips and I have a gap in my sentry/recent video footage - while the car was parked for an hour.  Any idea what might have caused this - as I was under the impression sentry would continuously record while parked?  Would the footage have been captured in the snapshots?

marcone, 2020-08-12:
I fixed the issue with the "snap--00001" folder being created.

closed 2020-08-12 by marcone