marcone/teslausb #501: musicarchive mountpoint looks corrupt

Vipercat, 2020-06-29:
Hi Marcone,

I have a question on one of the mountpoints (musicarchive) that looks corrupt/invalid as soon as the RPI reaches the point of 'Waiting for archive to be unreachable...' (see below, line is marked):

[image]

Any idea what is happening here?

Thanks,
Vipercat

marcone, 2020-06-29:
Never seen that before. Looks like the connection to the archive share is bad somehow (the share is normally only mounted while archiving, and gets unmounted afterwards)

Vipercat, 2020-06-29:
When I boot the RPI all seems fine until at a certain point the mountpoint turns to this state. I also noticed in the car that the CAM and MUSIC folders keep showing up and disappearing several times after an initial start. Then after half a dozen times (5 to 10 minutes) they stay mounted to the CAR. I will dig into the archiveloop.log and see if I can see something out of the ordinary.

Vipercat, 2020-06-30:
Hi Marcone,

I've checked my archiveloop.log and all I can see that might be not as it's should be, is that I don't see the /mnt/musicarchive unmounted, only the /mnt/archive (see end of log below):

Tue 30 Jun 02:01:09 CEST 2020: Starting archiveloop at 19.32 seconds uptime...
Tue 30 Jun 02:01:09 CEST 2020: Running fsck on /backingfiles/cam_disk.bin...
Tue 30 Jun 02:01:10 CEST 2020: | fsck from util-linux 2.33.1
Tue 30 Jun 02:01:17 CEST 2020: | fsck.fat 4.1 (2017-01-24)
Tue 30 Jun 02:01:17 CEST 2020: | /dev/loop0p1: 4 files, 3/2096063 clusters
Tue 30 Jun 02:01:17 CEST 2020: Finished fsck on /backingfiles/cam_disk.bin.
Tue 30 Jun 02:01:18 CEST 2020: Running fsck on /backingfiles/music_disk.bin...
Tue 30 Jun 02:01:18 CEST 2020: | fsck from util-linux 2.33.1
Tue 30 Jun 02:01:20 CEST 2020: | fsck.fat 4.1 (2017-01-24)
Tue 30 Jun 02:01:20 CEST 2020: | /dev/loop0p1: 2 files, 1/1057892 clusters
Tue 30 Jun 02:01:20 CEST 2020: Finished fsck on /backingfiles/music_disk.bin.
Tue 30 Jun 02:01:20 CEST 2020: Connecting usb to host...
Tue 30 Jun 02:01:20 CEST 2020: Connected usb to host.
Tue 30 Jun 02:01:20 CEST 2020: Waiting for archive to be reachable...
Tue 30 Jun 02:01:24 CEST 2020: Archive is reachable.
Tue 30 Jun 02:01:24 CEST 2020: Trying to set time...
Tue 30 Jun 02:04:23 CEST 2020: Time adjusted by 178.692294 seconds after 0.740000 seconds
Tue 30 Jun 02:05:23 CEST 2020: Disconnecting usb from host...
Tue 30 Jun 02:05:24 CEST 2020: Disconnected usb from host.
Tue 30 Jun 02:05:24 CEST 2020: Running fsck on /backingfiles/cam_disk.bin...
Tue 30 Jun 02:05:24 CEST 2020: | fsck from util-linux 2.33.1
Tue 30 Jun 02:05:26 CEST 2020: | fsck.fat 4.1 (2017-01-24)
Tue 30 Jun 02:05:26 CEST 2020: | /dev/loop0p1: 4 files, 3/2096063 clusters
Tue 30 Jun 02:05:26 CEST 2020: Finished fsck on /backingfiles/cam_disk.bin.
Tue 30 Jun 02:05:26 CEST 2020: Running fsck on /backingfiles/music_disk.bin...
Tue 30 Jun 02:05:26 CEST 2020: | fsck from util-linux 2.33.1
Tue 30 Jun 02:05:27 CEST 2020: | fsck.fat 4.1 (2017-01-24)
Tue 30 Jun 02:05:27 CEST 2020: | /dev/loop0p1: 2 files, 1/1057892 clusters
Tue 30 Jun 02:05:27 CEST 2020: Finished fsck on /backingfiles/music_disk.bin.
Tue 30 Jun 02:05:27 CEST 2020: Archiving...
Tue 30 Jun 02:05:27 CEST 2020: Ensuring cam archive is mounted...
Tue 30 Jun 02:05:27 CEST 2020: Mounting /mnt/archive...
Tue 30 Jun 02:05:29 CEST 2020: Mounted /mnt/archive.
Tue 30 Jun 02:05:29 CEST 2020: Ensured cam archive is mounted.
Tue 30 Jun 02:05:29 CEST 2020: Ensuring music archive is mounted...
Tue 30 Jun 02:05:29 CEST 2020: Mounting /mnt/musicarchive...
Tue 30 Jun 02:05:29 CEST 2020: Mounted /mnt/musicarchive.
Tue 30 Jun 02:05:29 CEST 2020: Ensured music archive is mounted.
Tue 30 Jun 02:05:29 CEST 2020: Checking saved folder count...
Tue 30 Jun 02:05:29 CEST 2020: Ensuring cam file is mounted...
Tue 30 Jun 02:05:29 CEST 2020: Mounting /mnt/cam...
Tue 30 Jun 02:05:29 CEST 2020: Mounted /mnt/cam.
Tue 30 Jun 02:05:29 CEST 2020: Ensured cam file is mounted.
Tue 30 Jun 02:05:33 CEST 2020: There are 0 event folder(s) with 0 file(s) to move.
Tue 30 Jun 02:05:33 CEST 2020: Trimming free space in /mnt/cam, which has 4 extents
Tue 30 Jun 02:05:33 CEST 2020: Trim complete, image now has 4 extents
Tue 30 Jun 02:05:33 CEST 2020: Unmounting /mnt/cam...
Tue 30 Jun 02:05:34 CEST 2020: Unmounted /mnt/cam.
Tue 30 Jun 02:05:34 CEST 2020: Finished archiving.
Tue 30 Jun 02:05:34 CEST 2020: Copying music...
Tue 30 Jun 02:05:34 CEST 2020: Starting music sync...
Tue 30 Jun 02:05:34 CEST 2020: Ensuring music backing file is mounted...
Tue 30 Jun 02:05:34 CEST 2020: Mounting /mnt/music...
Tue 30 Jun 02:05:34 CEST 2020: Mounted /mnt/music.
Tue 30 Jun 02:05:34 CEST 2020: Ensured music drive is mounted.
Tue 30 Jun 02:05:34 CEST 2020: Copying music from archive...
Tue 30 Jun 03:10:36 CEST 2020: Copied 369 music file(s), deleted 0, skipped 0 previously-copied files, encountered 0 co$Tue 30 Jun 03:10:36 CEST 2020: Sending Pushover message for moved files.
Tue 30 Jun 03:10:37 CEST 2020: Trimming free space in /mnt/music, which has 4 extents
Tue 30 Jun 03:10:38 CEST 2020: Trim complete, image now has 3 extents
Tue 30 Jun 03:10:38 CEST 2020: Unmounting /mnt/music...
Tue 30 Jun 03:10:38 CEST 2020: Unmounted /mnt/music.
Tue 30 Jun 03:10:38 CEST 2020: Finished copying music.
**Tue 30 Jun 03:10:39 CEST 2020: unmounting /mnt/archive**
Tue 30 Jun 03:10:39 CEST 2020: Connecting usb to host...
Tue 30 Jun 03:10:39 CEST 2020: Connected usb to host.
Tue 30 Jun 03:10:39 CEST 2020: Waiting for archive to be unreachable...

Any clue to why that is or is this by design?

Thanks,
Vipercat

marcone, 2020-06-30:
Looking at my own archiveloop.log I'm seeing this too. Sometimes it logs that it unmounted /mnt/musicarchive and sometimes it doesn't. Last night it didn't for example, but it's no longer mounted now, so it must have gotten unmounted somehow.
Not sure what's going on here.

Vipercat, 2020-06-30:
Which script should execute the unmount command for /mnt/musicarchive?

marcone, 2020-06-30:
disconnect-archive.sh

Vipercat, 2020-06-30:
Last question on this:

What does the line **if [ -e "$MUSIC_ARCHIVE_MOUNT" ]** do in disconnect-archive.sh?

I've removed this line from disconnect-archive.sh, not knowing the consequence because I have in now way the knowledge you have on scripting this solution, but it does seem to do the trick.

**"old" disconnect-archive.sh:**

 _#!/bin/bash -eu

# unmount the archive. Without this, the archive mounts can get into a
# state where the archive is reachable via the network, appears to be
# mounted, but the mount is inoperable and any attempt to access it
# results in a "host is down" message.
# Run this in the background, since unmounting can hang, which would
# block a return to archiveloop.

_{
  log "unmounting $ARCHIVE_MOUNT"
  if ! umount -f -l "$ARCHIVE_MOUNT"
  then
    log "unmount failed"
  fi

  if [ -e "$MUSIC_ARCHIVE_MOUNT" ]
  then
    log "unmounting $MUSIC_ARCHIVE_MOUNT"
    if ! umount -f -l "$MUSIC_ARCHIVE_MOUNT"
    then
      log "unmount failed"
    fi
  fi
} &_

**"new" disconnect-archive.sh:**

_#!/bin/bash -eu

# unmount the archive. Without this, the archive mounts can get into a
# state where the archive is reachable via the network, appears to be
# mounted, but the mount is inoperable and any attempt to access it
# results in a "host is down" message.
# Run this in the background, since unmounting can hang, which would
# block a return to archiveloop.

{
  log "unmounting $ARCHIVE_MOUNT"
  if ! umount -f -l "$ARCHIVE_MOUNT"
  then
    log "unmount failed"
  fi

  log "unmounting $MUSIC_ARCHIVE_MOUNT"
  if ! umount -f -l "$MUSIC_ARCHIVE_MOUNT"
  then
    log "unmount failed"
  fi
} &_

marcone, 2020-06-30:
`if [ -e "$MUSIC_ARCHIVE_MOUNT" ]` checks if /mnt/musicarchive exists (it only exists if you configured a music drive)

jun3280net, 2020-07-04:
I'm sorry - but I've arrived at the same problem as well - and I'm not savvy with linux.  Any suggestions on how to fix this issue?  Thank you.

[image]

[image]

Snapshots at over 2000+?
[image]

Honestly - I don't even know how to use the snapshots - (is that via connecting to the Teslausb AP? - because I don't use that function.)

Archiveloop hasn't been updated since June30.  However, I can see on my CIFS folder, the video clips have been updating.
[image]

I'm trying "touch /tmp/archive_is_unreachable"  now - hopefully that does something.  I might just need to re-install from scratch.

marcone, 2020-07-04:
@jun3280net looks like you haven't upgraded in a while, so you could try upgrading before you do a complete reinstall:
```
setup-teslausb selfupdate
setup-teslausb upgrade
```

jun3280net, 2020-07-04:
Hi @marcone Happy 4th of July.  I had tried :
#Updating
sudo -i
/root/bin/remountfs_rw			 # remounts /root/bin so that it is RW
/root/bin/setup-teslausb selfupdate      # this updates the setup-teslausb script itself
/root/bin/setup-teslausb                 # run the updated setup script to update everything
reboot

it didn't help.  I ended up doing a fresh install - which also gave me some issues (probably because I haven't done it in so long).  Does doing a fresh install with the latest image (teslausb-20200122), get all of the latest updates, or do I have to run the aforementioned commands as well?

Also - I read that you implemented an "automount" feature.  How long should it normally take for Teslausb to load up in the car?

Thanks,

marcone, 2020-07-05:
> Does doing a fresh install with the latest image (teslausb-20200122), get all of the latest updates,

yes

> or do I have to run the aforementioned commands as well?

no
 
> Also - I read that you implemented an "automount" feature. How long should it normally take for Teslausb to load up in the car?

It depends on whether you're in wifi range or not, but expect about a minute on a Pi Zero.

DonC52, 2020-09-05:
I have noticed something on the "new" release of the system. As your are using losetup to point to the "disks" you want to run fsck it can take a few seconds for the losetup -d to actually release the /dev back to the system. I kind of wonder if is an issue that may be messing with the disks?

MarcFinns, 2020-10-08:
FYI  -I had the same corruption and i ended up reinstalling everything

marcone, 2021-03-09:
The only way I can reproduce the "corrupt" /mnt/musicarchive is by stopping my NAS while /mnt/musicarchive is mounted on the Pi for archiving and syncing. This eventually resolves itself when teslausb unmounts the music archive again.

closed 2021-03-09 by marcone